google/xls #850: Fuzz DSLX parser

meheff, 2023-02-01:
We should fuzz the DSLX parser to ferret out crashes and (to some extent) bad error messages.  We'd be interested in random samples which are not valid DSLX programs. One way to do this is to take an set of existing DSLX programs and do random mutations such as: token removal, token insertion, token swapping and token mutation. In no case should the parser crash and it should always return an error message pointing to somewhere in the input. It would be interesting if the source span of the error message could be checked as well. I don't think we can guarantee that the span includes the mutated point of the program, but perhaps some related check could be performed.

cdleary, 2023-08-12:
Documenting my procedure for running AFL on the DSLX interpreter on my local machine for the OSS repo. (There are other mechanisms we have available, but running AFL overnight showed a slew of nice things to fix before we reach for fancier / more persistent mechanisms.)

Started with:

`sudo apt-get install afl++`

Building the binary with AFL instrumentation:

```
CC=afl-gcc CXX=afl-g++ bazelisk build -c opt //xls/dslx:interpreter_main
```

Making a simple seed corpus (I realized after I probably should have at least stripped out comments and similar):

```
mkdir /tmp/fuzz-in/
cp xls/dslx/tests/*.x /tmp/fuzz-in/
cp xls/dslx/tests/errors/*.x /tmp/fuzz-in/
```

Running AFL concurrently to use a bunch of cores (in different mux panes):

```
mkdir ~/multifuzz
afl-fuzz -M fuzzer01 -t 1500 -i /tmp/fuzz-in -o ~/multifuzz/  bazel-bin/xls/dslx/interpreter_main @@
afl-fuzz -S fuzzer02 -t 1500 -i /tmp/fuzz-in -o ~/multifuzz/  bazel-bin/xls/dslx/interpreter_main @@
...
afl-fuzz -S fuzzer08 -t 1500 -i /tmp/fuzz-in -o ~/multifuzz/  bazel-bin/xls/dslx/interpreter_main @@
```

Watching with `afl-whatsup` from yet another pane:

```
afl-whatsup ~/multifuzz/
```

Minimizing the findings later:

```
mkdir /tmp/fuzz-min
for f in $(ls ~/multifuzz/*/crashes/id:*); do afl-tmin -i ${f} -o /tmp/fuzz-min/$(basename ${f}) -- bazel-bin/xls/dslx/interpreter_main  @@; done
```

Largely could just follow the guidance from https://afl-1.readthedocs.io/en/latest/fuzzing.html which was great!

cdleary, 2023-08-12:
And I should note for context -- this is all in addition to our fuzzing and checking-for-equivalence of *valid* DSL programs -- this is specifically to try to find crashes and internal errors in the frontend when weird inputs are provided. The typical fuzzer (in `xls/fuzzer`) continues in its normal fashion.