google/xls #851: Enhance C++ testing ergonomics of DSLX code

meheff, 2023-02-03:
DSLX test capability is limited due to the limited expressibility of the language relative to software languages. In particular, DSLX-based testing of procs is especially awkward. DSLX test features could be added/enhanced à la SystemVerilog or nontrivial testing could be shifted to another language like C++.  I'd advocate leaning on C++ to do complex testing to avoid feature bloat in DSLX. Below is an example of what a C++ based testbench could look like:

DSLX to test:

```
struct Foo {
  bar: u17,
  baz: s44
}

struct ProcState {
  a: bool,
  b: u32
}

proc MyProc {
  c0: chan<Foo> in;
  c1: chan<u42> out;
  ...
}
```

Spitballing what a C++ test might look like below. The generated header defines all the necessary types to make things easy.

```C++
#include "my_proc.h" // auto-generated header

MyProc proc;
Foo input = { .bar=DslxBits<17>(10), .baz=DslxSBits<42>(123) };
proc.GetChannel("c0").Write(input);
proc.Tick();
EXPECT_EQ(proc.GetChannel("c1").Read(), DslxBits<42>UBits(444));
EXPECT_TRUE(proc.GetChannel("c1").empty());
EXPECT_EQ(proc.GetState(), ProcState {.a=DslxBits<1>(true), .b=DslxBits<32>(44)};
```

It might be good to support some of the features in the IR-level Proc runtime [interpreter/proc_runtime.h](https://github.com/google/xls/blob/main/xls/interpreter/proc_runtime.h) like `TickUntilOutput`, `TickUntilBlocked`., etc.

meheff, 2023-02-03:
@cdleary FYI

allight, 2024-05-08:
Splitting this into DSLX type things and proc-jit-wrapper https://github.com/google/xls/issues/1402 and jit-wrapper improvements https://github.com/google/xls/issues/1403.

proppy, 2024-05-15:
@allight did add a C++ proc sample in https://github.com/hzeller/xls/blob/3d9871fe629ac2bf1636514509e08100e3a37a57/xls/jit/jit_wrapper_test.cc#L108-L167

@meheff  do we want to add C++ tests for all proc examples we have under https://github.com/google/xls/blob/main/xls/examples/ ?

for some of the simpler example my personal hunch is that `test_proc` are straightforward enough, maybe we should limit C++ tests when they provide a clear value? (i.e: sweeping tests), what do others think?

proppy, 2024-05-15:
https://github.com/google/xls/tree/main/xls/examples/matmul_4x4 could be a good candidate to demonstrate the new proc jit support added in 3d9871f.