google/xls #852: Support explicit narrowing of structs in func/proc/channel I/O boundary.

hongted, 2023-02-04:
If may be desirable to group and store data in particular struct.  That struct is then reused across multiple functions, channels, and procs.  A given function/proc/channel may not use all fields in said struct.  In the final netlist, it is desirable that unused fields are of no-cost.  Currently this relies on  either the XLS to synthesis tool to detect unused registers / wires and remove them.  

Dead code/signal elimination at the I/O boundary may be more difficult so this issue is to develop a mechanism where the user can explicitly mark certain fields as "not used" thereby creating a set of related structs.

Possible language features like inheritance, attributes, interfaces, traits, ... can be used to express this idea.  Goal of this is to allow the user to explicitly narrow structs to minimize resource use for funcs and procs.

Example the I/O for each function and proc can be narrowed, but won't because I/Os are preserved (without inlining).
```
struct UnifiedData {
  a : u32,
  b: u32,
  c: u32,
  d: u32,
}

// func0 only uses fields a and b
fn func0(s : UnifiedData) -> u32 {
  s.a + s.b
}

// func1 only uses c and d
fn func1(s: UnifiedData) -> u32 {
  s.c + s.d
}

// proc uses all fields, but each channel 
// need not be the full width.
proc foo {
  c0 : chan<UnifiedData> in;
  c1: chan<UnifiedData> in;

  c2: chan<u32> out;
  next() {
    x = recv(c0)
    y = recv(c1)

    send(out, func0(x) + func1(y) + x.d);
  }
}

}
```

Proposed feature would be to be able to create a family a related structs that can be shared.
```
struct UnifiedData {
  a : u32,
  b: u32,
  c: u32,
  d: u32,
}

// func0 only uses fields a and b via something akin to inheritance?
struct NarrowedData0  : UnifiedData { 
  a: u32,
  b: u32,
}
 

fn func0(s : NarrowedData) -> u32 {
  s.a + s.b
}

// func1 only uses c and d via an attribute?
#[unused_field(a, c)]
type NarrowedData1  = UnifiedData;

fn func1(s: NarrowedData1) -> u32 {
  s.c + s.d
}
```

grebe, 2023-03-23:
The case where empty struct/tuple serves as a completely narrowed struct is notable- it could be a way to have ports included conditionally, or more generally be the basis for an Option type.

hongted, 2023-05-19:
A related feature would be able to explicitly narrow the bitwidth of a field -- possibly with a check so that if a specific field of 32 bits a is known to be in range [0, 8], then only 4 bits are actually generated by the tool.