google/xls #853: Add CMake support for this project

KuiWei004, 2023-02-06:
1. cmake can export complete compile_command.json，by which we can use c++ lsp to read source code well.
2. For a mostly C++ project, using cmake can make it easy to integrate xls to other project.

cdleary, 2023-02-15:
We're not able to be able to spend the extra support effort to actively support two build systems at this time, but if folks want to make a wrapping repo that has google/xls as a submodule and supplies a CMakeLists for it we'd be happy to point at it from the readme.

Just from some searching it seems like there are some ways of getting a `compile_command.json` from a Bazel build but I've not tried them myself.

KuiWei004, 2023-02-16:
> but if folks want to make a wrapping repo that has google/xls as a submodule and supplies a CMakeLists for it we'd be happy to point at it from the readme.

Could you please give some points from the readme?
Thanks!

KuiWei004, 2023-02-16:
> We're not able to be able to spend the extra support effort to actively support two build systems at this time, but if folks want to make a wrapping repo that has google/xls as a submodule and [...]

Could you please give some points from the readme?
Thanks!

cdleary, 2023-02-27:
> Could you please give some points from the readme?

Sorry I think I was unclear -- I meant if somebody (not us) does make a CMake build flow for XLS in a repo I'd be happy to reference it in the repo. We do not have bandwidth to try to make one at this time, though.

I'll also CC @hzeller here who has recently gotten `compile_commands.json` out of Bazel flows for use in `clang-tidy` scripts, I saw there are several things that say they help get `compile_commands.json` from Bazel flows but I wasn't sure if any of them were easily workable, sounds like Henner knows!

cdleary, 2023-04-03:
With Henner's changes you can now build the repo, run `./dev_utils/make-compilation-db.sh` (which produces a `compile_commands.json`) and then `./dev_utils/run-chang-tidy-cached.sh` -- this is documented here and rendered docs are live: https://github.com/google/xls/blob/dda27aad857470109383d6e7b907b5067cf9208b/docs_src/tools.md?plain=1#L217

closed 2023-04-03 by cdleary