google/xls #854: Slow loop unrolling

j2kun, 2023-02-07:
Cf. https://github.com/google/fully-homomorphic-encryption/issues/36

A user reported an issue with a program which, when I simplify it to the code below, seems like it should loop-unroll successfully, but it times out at 15 minutes when building. Any idea why this would be timing out? At a glance it seems obvious that it will terminate. but perhaps the break statements are adding too many branches in the Z3 model?

```
#pragma hls_top
void sort(int list[10], char L, char R) {
  char i, j;
  int pv;
  int tmp;

  i = L;
  j = R;

  pv = list[(L + R) / 2];

#pragma hls_unroll yes
  for (char k = 0; k < 4; k++) {
#pragma hls_unroll yes
    for (char ki = 0; ki < 4; ki++) {
      if (list[i] < pv) break;
      i++;
    }

#pragma hls_unroll yes
    for (char kj = 0; kj < 4; kj++) {
      if (pv < list[j]) break;
      j--;
    }

    if (i >= j) break;

    tmp = list[i];
    list[i] = list[j];
    list[j] = tmp;

    i++;
    j--;
  }
```

spurserh, 2023-02-15:
I created the --z3_rlimit parameter to xlscc, now the Phazon team needs to add it to their build to fix the issue.

spurserh, 2023-02-15:
(I suggest --z3_rlimit 5000 as a default)

j2kun, 2023-02-15:
Thanks! Just working out some issues with bumping the commit internally (xls won't build in CI). Happy to close this issue.

closed 2023-02-15 by j2kun