google/xls #857: Support sign_ext in std.x

meheff, 2023-02-10:
Sign extending a value from one width to another is more annoying than it should be. Cases which are particularly annoying are if the to/from type are not signed and when the to/from types are typedefs. For example:

```
type x_t  = u8
type y_t  = u16

let x = x_t:0xf;
let signext_x = (x as s8) as s16;
```

It'd be nice to be able to simply do:

```
std::sign_ext(x, y_t:0)
```

Or if we supported types as parametric parameters: 

```
std::sign_ext<y_t>(x)
```

hongted, 2023-02-10:
Does the built-in signex solve the immediate issue?  https://google.github.io/xls/dslx_reference/#signex

A separate concern would be if the language is flexible enough so that the built-in signex be written in dslx without being a built-in.

allight, 2026-11-14:
Closing this issue as obsolete (accepted by @allight): `std::sign_ext` was removed from `xls/dslx/stdlib/std.x` and replaced by the type-checked builtin `widening_cast<T>(x)` (`xls/dslx/stdlib/std.x:1174`), which statically prevents narrowing casts. (Closed by Gemini)

closed 2026-11-14 by allight