google/xls #859: Naming of binaries currently very generic

hzeller, 2023-02-10:
The tutorial mentions binary names like `repl` or  `codegen_main`, which are pretty generic names.

For a future `bazel run :install` that puts it in `/usr/bin/` these should probably get names that reflect what they do. Similar to `xlscc`, using that prefix might already be good (e.g. `xlsrepl`, `xlscodegen`), in particular as it provides for `<TAB>`-completable discovery of the available names.

meheff, 2023-02-10:
That sounds reasonable to me.  I like the prefixing idea.

cdleary, 2023-02-15:
cc @proppy -- would a `bazel run :install` / prefixing our binaries name make conda packaging easier / any big changes? Sounds good to me as well just wanted to loop Johan in on the notion.

proppy, 2023-04-27:
+1 on prefixing the binary, I think ideally we'd have a `:dist` rule https://bazelbuild.github.io/rules_pkg/0.8.0/reference.html#pkg_tar rules that bundle up public binaries used for external distro.

proppy, 2023-04-27:
currently the conda packages bundle the following binaries:
```
xls/dslx/interpreter_main
xls/dslx/ir_convert/ir_converter_main
xls/tools/opt_main
xls/tools/codegen_main
xls/tools/proto_to_dslx_main
xls/contrib/xlscc/xlscc
```
I think I'd be in favor of the following names instead:
```
xls-dslx-run
xls-dslx-convert
xls-cc-convert
xls-ir-opt
xls-ir-codegen
```
and maybe we don't actually need to package `proto_to_dslx_main` ?

wdyt?

hzeller, 2023-04-27:
Looks good. Same prefix for tab-completability, using the public facing common dash separator instead of implementation detail underscore separator, also avoiding other implementation detail `_main` suffix.

Also binaries clearly indicate what they are for (dslx, ir, cc).

Good naming, let's make it happen.

proppy, 2023-04-28:
> `xls-dslx-run`

`xls-dslx-test` may be better, since the main usage is to run unittest.

cdleary, 2023-04-28:
I'm good with those, modulo the bikeshed that I'd like to keep underscores instead of switching to dashes -- IMO binaries are generally all underscores in our universe (e.g. for launching on Borg and similar) as we discussed in a code review, and unix in general doesn't have a consistent preference, so I'd like us to keep what we've been using. If there's sources to cite on dashes really being the correct and common way in unix environments I'm happy to read/consider, but I don't /think/ there's a strong case for it, and google3 doesn't tend to do it that I've seen.

We probably also want to consider that the `/tools` directory is a bit of a dumping ground which is part of why we were separating out `_main` suffixes (was useful to distinguish binary targets in a `bazel query`) -- we should probably factor all the libraries out of tools as part of this and declare it has to be packaged-esque binaries only in `tools/`.

meheff, 2023-04-28:
I agree to what @cdleary said. In addition to separating libraries and tools it is probably worth separating tools for users and internal tools. We have a fair number of random one-off tools which we might use for internal purposes but which are  very unlikely to be useful to end users and will just confuse people if they are scattered about the user-facing tools.

hzeller, 2023-05-01:
Well, I'd like to offer the counter-argument that the typical Unix binaries _are_ actually with dashes:

```bash
ls /usr/bin/*_* | wc -l ; ls /usr/bin/*-* | wc -l
```

The rare time you encounter binaries with underscores is if they have some technical substring in there like `foo-X86_64-bar`.

Underscores in binary names strongly hint to a potential user that they are not end-user invokables but probably some internal artifact of something (like if you see `x86_64-linux-gnu-as` with its underscore you assume that it is just something that is called in a linker script).

I think binaries with underscores would be highly surprising and make the product look unfinished. If we make such a choice now, it is harder to fix later.

(Underscores or dashes are both common in borg FWIW).

hzeller, 2025-06-10:
The more external users we have the more important it is to have some good naming at least for the distributed binaries [from the releases](https://github.com/google/xls/releases).

The current names are not.

Suggestions:

  * Remove `_main` at the end of binay names.
  * Use some prefix to be discoverable and TAB-completable. `interpreter` is very generic, but `xls-interpreter` or `dslx-interpreter` is easy findable.
  * Use dashes instead of underscores for the separators in distributed binaries. Underscores are a real eye-sore and look like the maintainers don't care about their product.

In a personal project experimenting with XLS, I [had to rename the downloaded binaries](https://github.com/hzeller/xls-convolve/blob/main/shell.nix) to stay sane :)

I like @proppy 's suggestion from https://github.com/google/xls/issues/859#issuecomment-1526114224

ericastor, 2025-06-10:
Naming being the hard problem that it is, it's no surprise that this is provoking strong opinions! "Underscores are a real eye-sore and look like the maintainers don't care about their product" is a bit harsh, don't you think? I agree underscores within a binary name are less typical in most Unix environments, though... your argument from 2023 is still strong.

We also should probably try not to let the underscore-vs-dash debate undermine our ability to pick better names.

It's my week on gardening duty, so I'll start discussing this a bit more internally!