google/xls #871: Z3 times out in mutual exclusion pass

meheff, 2023-02-28:
This is crasher_2023-02-27_0c7a.x.

```
package sample

file_number 0 "fake_file.x"

chan sample__x14(bits[116], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[1222], init={0x4000_0000}) {
  x4: bits[62] = literal(value=1537228672809129301, id=6, pos=[(0,29,22)])
  x7: bits[62] = bit_slice_update(x4, __state, __state, id=9, pos=[(0,32,34)])
  x2: bits[1222] = not(__state, id=4, pos=[(0,27,23)])
  bit_slice.10: bits[1] = bit_slice(x7, start=0, width=1, id=10)
  x5: bits[1] = uge(x2, x2, id=7, pos=[(0,30,24)])
  x8: bits[1] = sub(bit_slice.10, x5, id=11, pos=[(0,33,36)])
  x9: bits[1] = not(x8, id=12, pos=[(0,34,19)])
  literal.13: bits[1] = literal(value=0, id=13, pos=[(0,35,25)])
  x15: (token, bits[116]) = receive(__token, predicate=x9, channel_id=0, id=18)
  x10: bits[1] = dynamic_bit_slice(x8, literal.13, width=1, id=14, pos=[(0,35,24)])
  x16: token = tuple_index(x15, index=0, id=21, pos=[(0,40,24)])
  x18: bits[1] = and(x5, x5, id=24, pos=[(0,42,25)])
  x13: bits[1] = and(x5, x5, id=17, pos=[(0,38,25)])
  x6: bits[62] = reverse(x4, id=8, pos=[(0,31,21)])
  x26: bits[1] = or_reduce(x10, id=33)
  x3: (bits[1222], bits[1222], bits[1222]) = tuple(x2, x2, __state, id=5, pos=[(0,28,45)])
  tuple_index.19: token = tuple_index(x15, index=0, id=19)
  x21: token = after_all(__token, x16, id=27)
  literal.3: bits[1] = literal(value=1, id=3)
  x11: bits[62] = reverse(x7, id=15, pos=[(0,36,22)])
  x12: bits[62] = not(x4, id=16, pos=[(0,37,19)])
  literal.20: bits[32] = literal(value=0, id=20, pos=[(0,40,25)])
  literal.22: bits[32] = literal(value=1, id=22, pos=[(0,41,27)])
  x17: bits[116] = tuple_index(x15, index=1, id=23, pos=[(0,41,26)])
  x19: bits[1] = dynamic_bit_slice(x10, x8, width=1, id=25, pos=[(0,43,25)])
  x20: bits[1] = not(x18, id=26, pos=[(0,44,20)])
  x22: bits[1] = neg(x13, id=28, pos=[(0,46,20)])
  literal.29: bits[1] = literal(value=1, id=29, pos=[(0,47,41)])
  x23: bits[1223] = one_hot(x2, lsb_prio=true, id=30, pos=[(0,47,31)])
  x24: bits[62] = neg(x6, id=31, pos=[(0,48,19)])
  x25: bits[1] = neg(x9, id=32, pos=[(0,49,20)])
  x27: bits[18] = dynamic_bit_slice(x4, x26, width=18, id=34, pos=[(0,51,23)])
  x28: bits[1] = eq(x3, x3, id=35, pos=[(0,52,25)])
  after_all.36: token = after_all(__token, tuple_index.19, x21, id=36)
  next (after_all.36, x2)
}
```

Repro

```
opt_main sample.ir
ZZZZZZZZZZZz.....
```

Note this is after the change which reinstated rlimit: 80d2bf3c74b0e6606853320e7ec3e0a5eb780488

taktoa, 2023-03-06:
Here's a minimized reproducer, after running it through opt_main and removing some nodes by hand.

```
package sample

file_number 0 "fake_file.x"

chan sample__x14(bits[116], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")

top proc __sample__main_0_next(__token: token, __state: bits[1222], init={0x4000_0000}) {
  x4: bits[62] = literal(value=1537228672809129301, id=6, pos=[(0,29,22)])
  x7: bits[62] = bit_slice_update(x4, __state, __state, id=9, pos=[(0,32,34)])
  x2: bits[1222] = not(__state, id=4, pos=[(0,27,23)])
  bit_slice.10: bits[1] = bit_slice(x7, start=0, width=1, id=10)
  x5: bits[1] = uge(x2, x2, id=7, pos=[(0,30,24)])
  x8: bits[1] = sub(bit_slice.10, x5, id=11, pos=[(0,33,36)])
  x9: bits[1] = not(x8, id=12, pos=[(0,34,19)])
  x15: (token, bits[116]) = receive(__token, predicate=x9, channel_id=0, id=18)
  x16: token = tuple_index(x15, index=0, id=21, pos=[(0,40,24)])
  next (x16, x2)
}
```

The thing that stands out to me here is the `x2: bits[1222]`. I think the size of the bitvector is causing issues in Z3, and indeed when I change it to `x2: bits[12]` and update the state element correspondingly, Z3 no longer hangs. I need to think for a bit about how to best handle this. It might require a nontrivial analysis to find the best places to cut off the graph with a variable to exclude large bitvectors.

taktoa, 2023-03-06:
The `x2` was a bit of a red herring, it appears that the real issue is in `x7: bits[62] = bit_slice_update(x4, __state, __state)`. In particular, using `__state` (with such a wide width) as the start index of a `bit_slice_update` appears to hang Z3. A simple mitigation would be to slice the start index (with a bounds check) to a fixed but large enough width like 64 bits (I don't think we'll ever see a bitvector with more than 2^64 bits).

commit cca83feb (2023-03-07): Add bit width limit for start index in bit_slice_update in Z3 IR translator

Without this limit, it would hang in the fuzzer. In the future we may
want to explicitly slice the index and implement bounds checking.

Fixes #871

PiperOrigin-RevId: 514772743
closed 2023-03-07 by copybara-service[bot]