google/xls #877: tool --version should print a version number

proppy, 2023-03-03:
Currently it seems to print a placeholder string:
```
🍡 codegen_main --version
codegen_main
```
we should hook this to at the very least print the git commit hash.

cdleary, 2023-03-04:
We may need to add `stamp = True` to our binary targets in order to be able to grab this information somehow in the binary. Or maybe the release script should pass `--stamp` to the whole build? https://bazel.build/docs/user-manual#stamp

Probably warrants a little investigation on the best way to do it, because with build caching like we have in-house, if all the artifacts were stamped differently at every revision I imagine it would defy build caching of XLS-generated artifacts. (i.e. inside Goog is more sensitive to stamping than sensitivity level I'd expect for Bazel workspace users.)

proppy, 2026-10-28:
I think this is fixed now w/ the release tarballs:
```
xls-v0.0.0-10898-g8ddae8bf4-linux-x64 $ ./ir_converter_main --version
v0.0.0-10898-g8ddae8bf4
```

closed 2026-10-28 by proppy