google/xls #885: trace_fmt message should be prefixed by source code line number

meheff, 2023-03-04:
Example:

```
struct MyStruct {
  x: u32
}

#[test]
fn test_foo() {
  let _ = trace_fmt!("{:x}", MyStruct { x:u32:0xabcd });
  ()
}
```

The `:x` format preference is ignored:

```
bazel-bin/third_party/xls/dslx/interpreter_main foo.x --logtostderr --logtostderr
...
[ RUN UNITTEST  ] test_foo
I0304 09:41:05.729963 2497319 bytecode_interpreter.h:85] MyStruct {
I0304 09:41:05.729991 2497319 bytecode_interpreter.h:85]   x: u32:43981
I0304 09:41:05.730002 2497319 bytecode_interpreter.h:85] }
```

Rather than the `I034...` gibberish which refers to the C++ line number of the bytcode interpreter, it should say something like:

```
foo.x:28]  MyStruct {
foo.x:28]    x: u32:43981
foo.x:28]  }
```

Exact form is up for discussion, also whether or not to include a time stamp.

proppy, 2024-06-17:
I believe this was fixed as part of #1465 with 0115aeea566f0f208ab7ff94260ec4f98661f13d and e8d59343cf58027e27ce980fed08e436cf16b857

closed 2024-06-17 by proppy

hzeller, 2024-06-17:
Affirmative.