google/xls #896: Unhandled node for delay estimation: invoke

13agupta, 2023-03-13:
Hi, I have a fairly basic question regarding code generation for function invocations in XLS IR. I have a very simple file with some XLS IR in it, and it contains a function invocation:

```
package test

file_number 0 "fake_file.x"

fn __test__callee(x: bits[32], y: bits[32]) -> bits[32] {
  ret add.3: bits[32] = add(x, y, id=3, pos=[(0,1,4)])
}

fn __test__caller(x: bits[32], y: bits[32]) -> bits[32] {
  z: bits[32] = umul(x, y, id=6, pos=[(0,4,12)])
  ret invoke.7: bits[32] = invoke(z, z, to_apply=__test__callee, id=7, pos=[(0,5,8)])
}
```

When I try to perform code generation, using `__test__caller` as the top level function, I get the following error:
```
Check failed: ::absl::OkStatus() == (xls::RealMain(ir_path)) (OK vs. UNIMPLEMENTED: Unhandled node for delay estimation in delay model 'asap7': invoke.7: bits[32] = invoke(z: bits[32], z: bits[32], to_apply=__test__callee, id=7, pos=[(0,5,8)]))
```

I was wondering whether there was a way to get around this error by creating a separate module for `__test_callee`. If so, is there documentation on how I can do it? Thanks.

grebe, 2023-03-13:
One of the implicit assumptions of codegen is that function invocations have been inlined. This inlining is performed during IR optimization (performed via `opt_main`).

In [`xls/examples`](https://github.com/google/xls/blob/main/xls/examples/BUILD#L490) you'll see some examples where the bazel macro `xls_ir_verilog()` is used on targets that come out of `xls_dslx_opt_ir` (which runs `opt_main`). Other verilog targets come from `xls_dslx_verilog()`, which internally invokes ir conversion and `opt_main`. If you're running XLS manually, you'll want to do ir conversion (if writing DSLX/C++, but if you're writing IR directly no conversion needed)-> `opt_main` -> `codegen_main`.

cdleary, 2023-03-13:
This is also an artifact of the fact the XLS model currently assumes you'll generate individual "leaf" verilog modules and "stitch" them together in some broader environment. We don't currently have a solid reference stitching flow, since lots of teams approach stitching differently. That's been on our todo list to help newcomers get started with a solution, something like Amaranth, which would help with integration with OSS FPGA development environments like LiteX. cc @proppy

cdleary, 2023-03-16:
Let us know if you still are seeing issues after running inlining. Closing for now.

closed 2023-03-16 by cdleary