google/xls #899: Named sends and receives for IOConstraints

grebe, 2023-03-17:
Codegen supports scheduling constraints between (channel_name, send/recv) pairs. If there is only one send or receive to a channel (which is currently a requirement for codegen), this makes sense. However, for II>1, you could have multiple sends and receives to the same channel and want to independently constraint them. There should be a mechanism to name send/recvs (or tokens?) and introduce scheduling constraints between the named entities.

Example: a proc is talking to a memory with `request`, `read_response`, and `write_completion` channels. You want to have a constraint between a write request and write completion, and a separate constraint between a read request and read response, but both requests happen on the same channel so you end up overconstrained (and, if the requests aren't mutually exclusive, potentially infeasible).

cdleary, 2023-04-03:
The individual operations have names in the IR, right? So wouldn't it be something like a codegen flag taking the node name? (We should ideally have source mapping info that lets us automatically determine IR node names from source positions as well as a bidirectional mapping.)

grebe, 2023-04-04:
Yeah, although it can be annoying to update flags when the source changes (and be very hard to keep node names up to date). If IO constraints could be added to the IR, node->node constraints could be added directly by the frontend for each RAM operation.

grebe, 2023-04-04:
Related: #911