google/xls #912: check_ir_equivalence gets some questions

intchar-hub, 2023-03-23:
When my inputs of check_ir_equivalence_main are two same ir files, however get the result: **slover result: satisfiable false**
```
root@DESKTOP-NBI1920:/mnt/g/xls# cat test.ir
package my_package

file_number 1 "./test.c"

top fn test(a: bits[32]) -> bits[32] {
  literal.2: bits[32] = literal(value=3, id=2, pos=[(1,5,5)])
  ret add.3: bits[32] = add(a, literal.2, id=3, pos=[(1,5,5)])
}

root@DESKTOP-NBI1920:/mnt/g/xls# cat test.opt.ir
package my_package

file_number 1 "./test.c"

top fn test(a: bits[32]) -> bits[32] {
  literal.2: bits[32] = literal(value=3, id=2, pos=[(1,5,5)])
  ret add.3: bits[32] = add(a, literal.2, id=3, pos=[(1,5,5)])
}

root@DESKTOP-NBI1920:/mnt/g/xls# ./bazel-bin/xls/tools/check_ir_equivalence_main test.ir test.opt.ir
Solver result; satisfiable: false
```

cdleary, 2023-03-23:
Right, the solver tries to answer the following "satisfiability" question:

  satisfy the equation, for some input X, `program0(X) != program1(X)`, what is X?

If there is no such X, the solver says, "sorry, I can't satisfy your question with any X".

Does that make sense? (Thanks for filing -- we could make the output text better here to explain this.)

intchar-hub, 2023-03-28:
Very thanks for your explanation.