google/xls #920: Last register inserted for the control validity using CodegenOptions::valid_control is unused

grotival, 2023-04-04:
When pipelining + using CodegenOptions::valid_control, a register is inserted for the last stage but is unused. This raises a linter error:
```
Signal 'pN_valid' is not read from in module xxx
```

cdleary, 2023-04-04:
@hongted I guess this would an enhancement to add a codegen option that quashes the last register if users don't care about reading the idle signal outside?

hongted, 2023-04-04:
That's one option, but I'd like to know more about the use-case.  The valid signal should be associated with the data signal and if the valid signal is not read, then what is being used to determine if the data is valid or not?

We do have plans to improve our unused register removal optimizations and to create a lint-waiver flow, but I feel that in this case this lint is actually pointing to a design issue.

hongted, 2023-04-04:
See https://github.com/google/xls/blob/main/xls/codegen/testdata/pipeline_generator_test_ValidSignal.svtxt for an example. 

The last stage p1_valid is set to out_valid which should be read by the outside block before ever using the out signal.

grotival, 2023-04-04:
I think the problem is when the in_valid is set but out_valid is std::nullopt.
then the last register in the last stage is never read for internal usage nor in the output wires.

this test case but without the second arg https://github.com/google/xls/blob/main/xls/codegen/pipeline_generator_test.cc#L663

hongted, 2023-04-04:
Ack -- I'm thinking this is best handled by a separate dead-code elimination pass which removes unused registers/op after codegen.  I'll turn this issue into that request.

We have a similar request in https://github.com/google/xls/issues/762

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/codegen/codegen_pass_pipeline.cc:100-120` runs `RegisterCombiningPass`, `IdentityRemovalPass`, `CsePass`, `BasicSimplificationPass`, and `DeadCodeEliminationPass` (plus `BlockConversionDeadTokenRemovalPass` in `xls/codegen/block_conversion_dead_token_removal_pass.cc`), eliminating unused pipeline valid registers and dead nodes after block conversion. (Closed by Gemini)

closed 2026-11-14 by allight