google/xls #924: Memory data dependencies should add token dependencies

grebe, 2023-04-05:
Right now, the following C++

```
mem[idx] = 3;
const auto val = mem[idx];
```

could allow the read to be scheduled before the write. Memory accesses get lowered to sends and receives, so there is no explicit data dependency. There is a write completion channel, but it returns an empty tuple.

A pass similar to the data dependency pass should be added that adds token dependencies between operations on the same memory. These dependencies need not be added if the addresses can be proven to not be equal.

cdleary, 2023-04-07:
Is this something the xlscc frontend would introduce or something you're expecting would be in a general XLS IR pass?

grebe, 2023-04-08:
An IR pass like the current data dependency pass.

grebe, 2023-04-11:
Yeah, that's the pass I meant.

My understanding is that the XLSCC frontend keeps things simple by not worrying about token dependencies, instead using the proc's token param for every operation that takes a token input and maintaining a list of tokens that are outputs of operations and putting them in an afterall for the proc's next tick. The XLSCC frontend does not track token relationships as they are not explicit in the source. Instead, the TokenDependencyPass infers the token structure from data dependencies.

I can see an argument for having the XLSCC frontend track token dependencies and emit IR that has the right token graph from the beginning, although I think it would amount to something fairly similar to the existing TokenDependencyPass. @spurserh, can you comment on the tradeoffs of having the frontend manage the token graph? I could imagine the frontend doing exactly what it does right now and running a token dependency pass before it dumps IR to a file. 

Memories throw a wrench in this because writes, as implemented in the IR, don't create a data dependency. A write is a send on a write request channel and a receive on a write completion channel which has a type of empty tuple. Write completion doesn't create any data dependencies.

The analysis to determine token relationships between memory operations could in general be tricky and perhaps is not suitable for the frontend. The frontend could be conservative and make all memory operations sequential, but you could imagine it being important that e.g. a read and write to different addresses be schedulable in the same cycle. Figuring out that the addresses are different could be involved. Also, I think we're much more comfortable adding constraints to the token graph- removing them is tricky. If the frontend emits a fully sequential token graph, is it OK for analysis to make it more parallel? We don't have a way of distinguishing between "this is what I think the dependencies are, but if I'm wrong go nuts" from "there's side-effecting behavior not contemplated in the model you use for analysis (e.g. another proc is reading the memory) so don't remove this edge".

cdleary, 2023-04-12:
Thanks for the explanation -- I do think this could use a little writeup as the strategy is nailed down -- I feel like it falls into the bucket of "maybe XLScc could do with its own pre-processing pass pipeline that's specific to XLScc". This wouldn't be necessary for a frontend that managed token dependencies like DSLX, if I'm understanding correctly?