google/xls #926: `fail!()` in proc gives internal error

cdleary, 2023-04-07:
Reported by mtikekar@

https://groups.google.com/g/xls-dev/c/LgPOsMdDG2w/m/vi2ty7idBgAJ?utm_medium=email&utm_source=footer

```dslx
fn out_fn(x: u32, state: u32) -> u32 {
  match x {
    u32:0 => fail!("bad_x", state),
    _ => x + state
  }
}
proc Fmac {
  input_a_consumer: chan<u32> in;
  output_producer: chan<u32> out;

  init { u32:0 }

  config(input_a_consumer: chan<u32> in, output_producer: chan<u32> out) {
    (input_a_consumer, output_producer)
  }

  next(tok: token, state: u32) {
    let (tok, input_a) = recv(tok, input_a_consumer);
    let result = out_fn(input_a, state);
    let tok = send(tok, output_producer, result);
    result
  }
}
```

hongted, 2023-04-07:
Looks like this is related to https://github.com/google/xls/issues/862 .

mtikekar, 2023-04-07:
FYI, with the inlining pass disabled, it hit -

https://github.com/google/xls/blob/00ad2b68c5303984fafbae6430af6d6fd859c1c0/xls/passes/token_provenance_analysis.cc#L52

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): `assert!` and `fail!` inside functions called by procs are supported via `CallingConvention::kImplicitToken` and `FunctionConverter::HandleFailBuiltin` in `xls/dslx/ir_convert/function_converter.cc:2515-2543`, which thread implicit tokens and control predicates through helper functions into procs. (Closed by Gemini)

closed 2026-11-14 by allight