google/xls #934: quickcheck should support testing function w/ `match` expressions

proppy, 2023-04-20:
It seems that quickcheck doesn't currently resolve function with `match` expression, see the following example:
```
fn match1(a: u32, b: u32) -> u32 {
    match(a == b) {
        true => u32:0,
        false => u32:1,
        _ => fail!("unmatched_cond", s32:-1 as u32),
    }
}

#[quickcheck(test_count=50000)]
fn prop_match(a: u32, b: u32) -> bool {
    match1(a, b) == u32:1
}
```
which produce the following error:
```
E0420 08:34:30.630663      16 run_routines.cc:379] Internal error: NOT_FOUND: Package does not have a function with name: "__gcd__prop_match"; available: [____std__iterative_binary_div__64_32_counted_for_0_body, __std__iterative_binary_div__64_32, ____gcd__gcd_euclidean__32_counted_for_0_body, __gcd__gcd_euclidean__32, __itok__gcd__gcd_binary_match__32, ____itok__gcd__gcd_binary__32_counted_for_0_body, __itok__gcd__gcd_binary__32, __itok__gcd__prop_gcd_equal, __itok__gcd__match1, __itok__gcd__prop_match]
[                FAILED ] prop_match: internal error: NOT_FOUND: Package does not have a function with name: "__gcd__prop_match"; available: [____std__iterative_binary_div__64_32_counted_for_0_body, __std__iterative_binary_div__64_32, ____gcd__gcd_euclidean__32_counted_for_0_body, __gcd__gcd_euclidean__32, __itok__gcd__gcd_binary_match__32, ____itok__gcd__gcd_binary__32_counted_for_0_body, __itok__gcd__gcd_binary__32, __itok__gcd__prop_gcd_equal, __itok__gcd__match1, __itok__gcd__prop_match]
```

cdleary, 2023-04-27:
Quickcheck not handling functions that have `fail!` (which on IR conversion have a token threaded through them since assertions are side-effects). Also note that this sample wants #204

proppy, 2024-03-05:
Related #1110

closed 2024-03-05 by copybara-service[bot]