google/xls #939: assert_eq() not available under #[test_proc]

Abhiroop25902, 2023-04-26:
Code which was giving error (D Flip Flop with testing):
```
proc dff {
  clk_consumer: chan<u1> in;
  d_consumer: chan<u1> in;
  q_producer: chan<u1> out;

  init { (u1:0b0) }

  config(clk_consumer: chan<u1> in, d_consumer: chan<u1> in, q_producer: chan<u1> out) {
    (clk_consumer, d_consumer, q_producer)
  }

  next(tok: token, state: u1) {
    let (tok_clk, clk) = recv(tok, clk_consumer);
    let (tok_d, d) = recv(tok, d_consumer);
    let new_state = if clk == u1:0b1 {d} else {state};
    let tok = join(tok_clk, tok_d);
    let tok = send(tok, q_producer, state);

    new_state
  }
}

#[test_proc]
proc dff_test {
    input_dff_clk_sender: chan<u1> out;
    input_dff_d_sender: chan<u1> out;
    output_dff_q_receiver: chan<u1> in;
    terminator: chan<bool> out;

    init { () }

    config(terminator: chan<bool> out) {
        let (input_dff_clk_sender, input_dff_clk_receiver) = chan<u1>;
        let (input_dff_d_sender, input_dff_d_receiver) = chan<u1>;
        let (output_dff_q_sender, output_dff_q_receiver) = chan<u1>;
        spawn dff(input_dff_clk_receiver, input_dff_d_receiver, output_dff_q_sender);
        (input_dff_clk_sender, input_dff_d_sender, output_dff_q_receiver, terminator)
    }

    next(tok: token, state: ()) {
        let tok = send(tok, input_dff_clk_sender, u1:0b0);
        let tok = send(tok, input_dff_d_sender, u1:0b0);
        let (tok, q) = recv(tok, output_dff_q_receiver);
        let _ = assert_eq(q, u1:0b0);

        let tok = send(tok, terminator, true);
        ()
    }
}
```

The error being thrown:
```
user_module.x:46:26-46:37
0044:         let tok = send(tok, input_dff_d_sender, u1:0b0);
0045:         let (tok, q) = recv(tok, output_dff_q_receiver);
0046:         let _ = assert_eq(q, u1:0b0);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^---------^ ConversionError: Could not find name for invocation: assert_eq; available: [dff.init, dff.config, dff.next, dff_test.init, dff_test.config, dff_test.next]
0047: 
0048:         let tok = send(tok, terminator, true);
```

Enviornment Detail: added the D Flip Flop Code in [this](https://colab.research.google.com/github/chipsalliance/silicon-notebooks/blob/main/xls-adder-openlane.ipynb#scrollTo=JKGxScUtoV4E) google colab notebook and receiving this error.

proppy, 2023-04-26:
I think this is might be a duplicate of #673?

proppy, 2023-04-26:
I also hit this recently, according to @cdleary, currently `test_proc` requires to `interpreter_main` to be invoked with the following flag `--compare=none` to avoid the comparison against intepreted and jitted IR (see #673)

cdleary, 2023-04-26:
Yeah, appreciate you reporting this! If it's going to stay this way we should make the default better automatically, but even better would be if we just fix the conversion. :-)

Abhiroop25902, 2023-04-27:
Flag `--compare=none` seemed to work, thanks a lot for the heads up.

closed 2023-04-27 by Abhiroop25902