google/xls #944: build error python 3.10

proppy, 2023-05-01:
It seems that some of the dependencies don't resolve for more recent versions of python (ex: `3.10`), similar to what was highlighted in https://github.com/google/xls/issues/849#issuecomment-1475359010.

```
2023-05-01T04:52:59.7766932Z Collecting pyyaml==5.4.1
2023-05-01T04:52:59.7767247Z   Downloading PyYAML-5.4.1.tar.gz (175 kB)
2023-05-01T04:52:59.7767667Z      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 175.1/175.1 KB 53.1 MB/s eta 0:00:00
2023-05-01T04:52:59.7767935Z   Installing build dependencies: started
2023-05-01T04:52:59.7768316Z   Installing build dependencies: finished with status 'error'
2023-05-01T04:52:59.7768681Z  (  error: subprocess-exited-with-error
2023-05-01T04:52:59.7768903Z   
2023-05-01T04:52:59.7769373Z   × pip subprocess to install build dependencies did not run successfully.
2023-05-01T04:52:59.7769705Z   │ exit code: 1
2023-05-01T04:52:59.7769946Z   ╰─> [2 lines of output]
2023-05-01T04:52:59.7770275Z       ERROR: Could not find a version that satisfies the requirement setuptools (from versions: none)
2023-05-01T04:52:59.7770642Z       ERROR: No matching distribution found for setuptools
2023-05-01T04:52:59.7770886Z       [end of output]
```

This version of pyyaml (`5.4.1`)  predate python 3.10 release, so I guess that updating to `6.0.0` should fix this issue?

cdleary, 2023-10-31:
@proppy can we close this now that the build is grabbing its own Python?

proppy, 2023-10-31:
yep! + the given dep has been updated: https://github.com/google/xls/blob/78b254c45a43aaf66eef233d7f11e31a7767799f/dependency_support/pip_requirements.in#L10

closed 2023-10-31 by proppy