google/xls #947: binding error access tok from for expressions body

proppy, 2023-05-02:
It seems that `token` are scoped differently from other `let` bindings.

The following code:
```
import std

proc counter {
  output: chan<u8> out;

  init {
    u8:0
  }

  config(output: chan<u8> out) {
    (output,)
  }

  next(tok: token, state: u8) {
    let tok = send(tok, output, state);
    state + u8:1
  }
}

#[test_proc]
proc counter_test {
  terminator: chan<bool> out;
  output_s: chan<u8> out;
  output_r: chan<u8> in;

  init {
    ()
  }

  config(t: chan<bool> out) {
    let (output_s, output_r) = chan<u8>;
    spawn counter(output_s);
    (t, output_s, output_r)
  }

  next(tok: token, state: ()) {
    let increment = u8:1;
    let counter = for (i, expected) in u32:0..u32:32 {
      let _ = assert_eq(increment, u8:1);
      //let _ = assert_eq(state, ());
      let (tok, value) = recv(tok, output_r);
      let _ = assert_eq(value, expected);
      value + increment
    }(u8:0);
    let tok = send(tok, terminator, true);
    ()
  }
}
```
would produce the following error:
```
E0502 09:15:45.821443   18125 command_line_utils.cc:46] Could not extract a textual position from error message: INTERNAL: Could not find slot or binding for name: tok @ apu_pulse.x:39:8-39:11: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: Could not find slot or binding for name: tok @ apu_pulse.x:39:8-39:11
```
while the `increment` `let` binding seems be accessible from the previous `assert_eq` statement:
```
      let _ = assert_eq(increment, u8:1);
```
interestingly uncommenting:
```
      //let _ = assert_eq(state, ());
```
seems to workaround the issue and make the `tok` binding visible from the `for` expression body:
```
[ RUN UNITTEST  ] counter_test
[            OK ]
[===============] 1 test(s) ran; 0 failed; 0 skipped.
```

proppy, 2023-05-02:
this naturally go away when scoping the token appropriatly:
```
  next(tok: token, state: ()) {
    let increment = u8:1;
    let (counter, tok) = for (i, (expected, tok)) in u32:0..u32:32 {
      let _ = assert_eq(increment, u8:1);
      let (tok, value) = recv(tok, output_r);
      let _ = assert_eq(value, expected);
      (value + increment, tok)
    }((u8:0, tok));
    let tok = send(tok, terminator, true);
    ()
  }
```
but it still seems a confusing behavoir to scope `token` parameters differently from other `let` bindings.

cdleary, 2023-05-20:
Note that any internal error (e.g. when it says "Could not extract a textual position from error message:") means that the system did not behave properly in parsing / type checking / conversion to XLS IR. So this is a bug -- tokens should not have any different scoping properties from other identifiers as you say.

cdleary, 2023-05-20:
I think I coincidentally fixed this recently, perhaps via https://github.com/google/xls/commit/45ebdcba10922ecfa66e1e2009dc5a74d01e0617 -- will add a regression test.

closed 2023-05-20 by copybara-service[bot]