google/xls #955: Support ops that need implicit tokens inside for loops.

hongted, 2023-05-08:
Placing ops that need implicit tokens inside a for loops results in an IR verificaiton error
```
Side-effecting token-typed nodes must be connected to the sink token value via a path of tokens
```

```
  let _ = for (i, res): (u32, ()) in u32:0 .. u32:1 {
    let _ = cover!("cover_C", i == u32:1);
    res
  }(());
 ```

Related to https://github.com/google/xls/issues/691

cdleary, 2023-08-18:
@grebe you just fixed this today I think in https://github.com/google/xls/commit/71a5d1e45aa4d66b00cbbf0257ac9750c9e33d82 but please confirm, thanks!

grebe, 2023-08-18:
Yeah, this should work. I didn't specifically test covers (just trace), but it should work.

closed 2023-08-18 by grebe