google/xls #959: [IR optimization] Over eager IR optimization that leads to incorrect graph

mtdudek, 2023-05-16:
DSLX allows for process to spawn subprocesses.
It also allows for `procs` to be parameterizable,
which leads to situations were user would like to specialize
`proc <T>A` by spawning its instance in `proc B`'s config
and leave `proc B`'s next function empty (Please see attached file chan_min.x). 

Converting chan_min.x to IR creates 2 processes, inner and outer,
outer is top level process. Outer's next function is empty,
while inner's next function performs some operations on IO channels.
`bazel-bin/xls/dslx/ir_convert/ir_converter_main --top="outer" --dslx_path=./ chan_min.x`
<details><summary>unoptimized IR</summary>

 ```
  package chan_min
  
  file_number 0 "../chan_min.x"
  chan chan_min___in(bits[32], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
  chan chan_min___out(bits[32], id=1, kind=streaming, ops=send_only, flow_control=ready_valid, metadata="""""")
  
  fn __chan_min__inner.init__32() -> () {
    param: bits[32] = literal(value=32, id=1, pos=[(0,0,15)])
    ret tuple.2: () = tuple(id=2, pos=[(0,3,11)])
  }
  
  top proc __chan_min__outer_0_next(__token: token, __state: (), init={()}) {
    literal.5: bits[1] = literal(value=1, id=5)
    tuple.6: () = tuple(id=6, pos=[(0,23,8)])
    after_all.7: token = after_all(__token, id=7)
    next (after_all.7, tuple.6)
  }
  
  proc __chan_min__outer__inner_0__32_next(__token: token, __state: (), init={()}) {
    receive.12: (token, bits[32]) = receive(__token, channel_id=0, id=12)
    tok: token = tuple_index(receive.12, index=0, id=14, pos=[(0,8,13)])
    data: bits[32] = tuple_index(receive.12, index=1, id=15, pos=[(0,8,18)])
    tuple_index.13: token = tuple_index(receive.12, index=0, id=13)
    tok__1: token = send(tok, data, channel_id=1, id=16)
    literal.10: bits[1] = literal(value=1, id=10)
    param: bits[32] = literal(value=32, id=11, pos=[(0,0,15)])
    tuple.17: () = tuple(id=17, pos=[(0,10,8)])
    after_all.18: token = after_all(__token, tuple_index.13, tok__1, id=18)
    next (after_all.18, tuple.17)
  }
```
</details>

Simulating unoptimized IR and DSLX side-by-side using interpreter passes without issues.
`bazel-bin/xls/dslx/interpreter_main --dslx_path=./ chan_min_test.x`

After optimizing IR graph, inner process and IO channels are removed,
which creates empty IR graph.
`bazel-bin/xls/tools/opt_main chan_min.ir`
<details><summary>Incorrectly optimized IR</summary>

```
package chan_min

file_number 0 "../chan_min.x"

top proc __chan_min__outer_0_next(__token: token, init={}) {
  after_all.7: token = after_all(__token, id=7)
  next (after_all.7)
}
```
</details>

If `--top` proc is forced to be the inner's next using
`bazel-bin/xls/tools/opt_main ../chan_min.ir --top=__chan_min__outer__inner_0__32_next`
optimized IR graph is correct.
<details><summary>Correctly optimized IR</summary>

```
package chan_min

file_number 0 "../chan_min.x"

chan chan_min___in(bits[32], id=0, kind=streaming, ops=receive_only, flow_control=ready_valid, metadata="""""")
chan chan_min___out(bits[32], id=1, kind=streaming, ops=send_only, flow_control=ready_valid, metadata="""""")

top proc __chan_min__outer__inner_0__32_next(__token: token, init={}) {
  receive.12: (token, bits[32]) = receive(__token, channel_id=0, id=12)
  tok: token = tuple_index(receive.12, index=0, id=14, pos=[(0,8,13)])
  data: bits[32] = tuple_index(receive.12, index=1, id=15, pos=[(0,8,18)])
  tok__1: token = send(tok, data, channel_id=1, id=16)
  after_all.18: token = after_all(__token, tok, tok__1, id=18)
  next (after_all.18)
}
```
</details>

It looks like `spawn` is lost during IR conversion, which causes IR graph to forget ,
that inner next process is an internal part of the outer `proc`.  
This later leads to the removal of the inner proc and IO channels.

One way to solve this issue would be to keep track, which proc has spawned other proc,
and only delete proc's going from youngest to oldest.
Other option is to track data flow through internal proc/chan graph, and keep all procs
that are on any path from input to output.

meheff, 2023-06-08:
Thanks for filing this. You've definitely identified a hole in the IR representation. In the frontend there is a heirarchy of procs, but this structure is discarded when lowered to IR where the only connectivity between procs is a set of global channels. The road to fixing this involves scoping channels to procs (https://github.com/google/xls/issues/869) which would mirror the representation in DSLX. Not so coincidentally that also mirrors the representation in verilog where you have a heirarchy of instantiated modules. Closing this as a duplicate of that issue as changing to proc-scoped channels will resolve this issue.

I have a document outlining how this might be done. I'll try to export it into our design doc folder.

closed 2023-06-08 by meheff

proppy, 2023-06-09:
https://github.com/google/xls/commit/bf28838aa0fc0e00ac6165511ffbc5e5f4430195 fyi