google/xls #965: [DLSX enhancements] Add `send_non_blocking` builtin function

mtdudek, 2023-05-17:
Right now DSLX proc's will always stall on `send`, if no other proc is actively waiting on `recv`.
This approach makes so no data will be lost/dropped, but it takes away control from users.

I propose adding  `send_non_blocking` and `send_if_non_blocking`. This way send
will not block pipeline and users will be able to decide, what they want to do with data, like:
* dropping it,
* buffering it,
* sending it over other channel,
* or something else.

These functions would have following signatures:
* `send_non_blocking(:token, :chan<:T>, :T) -> (:token,  :u1)`,
* `send_if_non_blocking(:token, :chan<:T>, :u1, :T) -> (:token, :u1)`.

Second element in the tuple is 0/1 flag. If transfer was successful flag
would be high, otherwise it would be low.

/cc @proppy

cdleary, 2023-05-20:
This is less of a DSLX specific thing, we'd have to figure out if we can add it to XLS' communicating process semantics. The DSL in many senses a "thin layer" on top of the XLS IR that does parametric instantiation and some other sugary things.

For context, generally we try to only add operations to the model that can still obey latency-insensitivity, i.e. with only these latency-insensitive ops, even with changes in pipeline scheduling decisions the overall design would tend to act the same and the same DV testbench could be used with no modifications. Certain non-blocking operations make the design latency sensitive, where you'd observe differences between unscheduled- and scheduled-execution of the same program. Arbiters are a classic example of where latency is observable in the resulting schedule and output. We try to adhere to a KPN-style model for procs, but unfortunately we don't have a good markdown doc on our model / semantics.

cc @grebe who recently wrote a design doc around proc transforms and @hongted who thought about how to add a peek() op and whether that could still be latency insensitive.

proppy, 2023-05-30:
@mtdudek my undestanding is that this is not blocking #994 and #995, can you confirm?

mtdudek, 2023-05-30:
This issue does not block #994 and #995 in anyway.