google/xls #968: [DSLX/IR] `xls_dslx_test` doesn't compare interpreter results with JIT

mtdudek, 2023-05-18:
PR #958 allowed for `compare=none` to be removed from `xls_dslx_test`s that are using test_proc.
One of such designs is `matmul_4x4`. `matmul_4x4_dslx_test` target is successful, but it shouldn't be.

IR generated from `matmul_4x4` is empty:
```
package matmul_4x4

```, so JIT and interpreter results should be different.

/cc @proppy

mtdudek, 2023-05-18:
I think it has something to do with
https://github.com/google/xls/blob/main/xls/dslx/run_comparator.cc#L52-L61
It looks like JIT will run only top level functions, but proc's next function is not a top level function.

After further investigation, it's connected with
https://github.com/google/xls/blob/6bc7853f6fc3a06b925abc84dd5ffc234df513b8/xls/dslx/ir_convert/extract_conversion_order.cc#L636-L656

I think that the code removed in #958, was there to gather functions/procs 
that are called in `#[test]` and `#[test_proc]` respectively, but due to the way
`AddToReady` was constructed I got the wrong idea( `AddToReady` was no-op for TestFunction).

I'm checking if only adding callees will be enough.

mtdudek, 2023-05-22:
Simply adding all test_proc.config callees was not enough to get JIT working.

This changed caused undefined variable to appear in callees' config functions.
Adding full test_proc solved undefined variables in callees config,
but produced issue with redefining `terminator` channel.

I tried generating IR for each test individually. It solved issue with symbols redefinition.
It lead to undefined symbols in the test_proc.next function if proc's variable was used in `for` statement.

I tried binding freevar from `for loop` statements, but got further issue with `send` and `recv` statements.
Currently DSLX to IR conversion expects loop bodies to be function only,
but channel operations aren't allowed in functions.

proppy, 2023-05-24:
> I think that the code removed in https://github.com/google/xls/pull/958, was there to gather functions/procs

IIRC our understanding (after reading the code together) was that it wasn't succesfull at doing so.

I think the initial reason behind comparing the ir interpreter/jit result, was to give an heads up to that dslx code was not convertible before calling `ir_converter_main`. It seems that non-convertible features being a known subset of the DSLX spec, this would be more efficiently achieved by flagging unimplemented keywords and expressions in the AST.

/cc @hongted @leary

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls_dslx_test` runs both `#[test]` functions and `#[test_proc]` procs through `DslxInterpreterParsedTestRunner` and `IrJitTestRunner` (`RunTestProc` in `xls/dslx/run_routines/ir_test_runner.cc:159,239` and `xls/dslx/run_routines/run_routines.cc:305,1108`) when `compare = "jit"` is enabled. (Closed by Gemini)

closed 2026-11-14 by allight