google/xls #981: DSLX: Misleading `assert_eq` failure message

rw1nkler, 2023-05-24:
I have noticed that when `assert_eq()` fails on a check that includes a state variable, the final error message is misleading.
Here is a small example showing the problem:
```
#[test_proc]
proc Test {
    terminator: chan<bool> out;
    
    init {(u32:0)}

    config(terminator: chan<bool> out) {
        (terminator,)
    }

    next(tok: token, counter: u32) {
        trace_fmt!("counter: {}", counter);
        let _ = assert_eq(counter < u32:10, true);
        counter + u32:1
    }
}
```

And the output:
```
 [ RUN UNITTEST  ] Test
I0524 16:47:25.402100 1264846 bytecode_interpreter.h:85] counter: 0
I0524 16:47:25.402162 1264846 bytecode_interpreter.h:85] counter: 1
I0524 16:47:25.402187 1264846 bytecode_interpreter.h:85] counter: 2
I0524 16:47:25.402191 1264846 bytecode_interpreter.h:85] counter: 3
I0524 16:47:25.402212 1264846 bytecode_interpreter.h:85] counter: 4
I0524 16:47:25.402216 1264846 bytecode_interpreter.h:85] counter: 5
I0524 16:47:25.402220 1264846 bytecode_interpreter.h:85] counter: 6
I0524 16:47:25.402225 1264846 bytecode_interpreter.h:85] counter: 7
I0524 16:47:25.402228 1264846 bytecode_interpreter.h:85] counter: 8
I0524 16:47:25.402233 1264846 bytecode_interpreter.h:85] counter: 9
I0524 16:47:25.402236 1264846 bytecode_interpreter.h:85] counter: 10
E0524 16:47:25.402281 1264846 run_routines.cc:274] Internal error: NOT_FOUND: Could not find concrete type for node: (counter) < (u32:10)
[        FAILED ] Test: internal error: NOT_FOUND: Could not find concrete type for node: (counter) < (u32:10)
[===============] 1 test(s) ran; 1 failed; 0 skipped.
```

cdleary, 2023-05-24:
Note that anything that includes "internal error" that is surfaced to the user is a bug.

mtdudek, 2023-05-26:
Hi,
I've opened PR #990 with a possible fix to this issue.

proppy, 2023-05-30:
@rw1nkler @mtdudek is that blocking the ability to write tests for #994 and #995?

mtdudek, 2023-05-30:
Asserts will work as intended, they will fail on error and exit test.

Output from such tests will be of zero value, as it will not give any inside 
why assertion failed, only that it failed.

We will be able to test working designs, and see that broken designs fail,
but we will not know why that design failed.

PR #990 (2023-07-19): [DSLX] Propagating type_info through proc frames
This commit fixes internal error when testing DSLX procs using `#test_proc()`.
Issue was that only first evaluated frame had `type_info` defined, and all following frames had it undefined. This made ConcreteType lookup fail.

Fixes #981
closed 2023-07-19 by copybara-service[bot]