google/xls #984: zero! builtin don't work with  parametric structs and arrays of structs.

proppy, 2023-05-26:
https://google.github.io/xls/dslx_reference/#zerot doesn't seems to work with parametric strucst and arrays of structs.

Is that a known limitation, are there possible workaround?

cdleary, 2023-05-26:
Re: #982 -- we generally don't want to introduce more backtracking into the parser as an implementation strategy, since it makes for bad error messages and slow parsing. I was hoping to restructure the grammar around types a bit, this is one of the first instances where we have parametric types so the grammar was a bit overfit for parametric-values-only.

The test cases looked good though, maybe those can be used with a DISABLED_ and a TODO referencing this issue?

See also related #904

cdleary, 2023-05-26:
Oh, and the workaround is to define a type alias and then use that as the parametric argument to `zero!`

mtdudek, 2023-05-26:
I've created PR #985 with tests and disabled ones that don't work right now.
I think #976  can benefit from adding type parsing to parametric pass.

I've also created draft PR #986 with a possible solution to parse parametric types.
It doesn't parse `::` after first `<` or `[`, it also doesn't allow for using
fields from const arrays, like:
```
const a = u32[2] : {1,2}

fn b<A:u32>(x: u32) -> u32 {
  x + A
}

b<a[0]>(a[1])
```
As far as I can tell, using const array's fields was never allowed, so proposed solution doesn't break it.

proppy, 2023-05-30:
@mtdudek is that blocking #994 and #995?

mtdudek, 2023-05-30:
It's not blocking. There is workaround to pass any type to zero!<> using type definitions.

mikex-oss, 2024-06-18:
+1. Just ran into this myself.

commit a2551542 (2024-07-18): Fix bug where zero!<MyType[3]>, zero!<MyType<X>>, etc. fail to parse.

https://github.com/google/xls/issues/984

Fixes #984

PiperOrigin-RevId: 653406922
closed 2024-07-18 by copybara-service[bot]