google/xls #1001: Delay model generation: add support for specialization HAS_LITERAL_OPERAND

tcal-x, 2023-05-30:
The proto-driven delay characterization scripts do not currently have support for the `HAS_LITERAL_OPERAND` specialization.

tcal-x, 2023-09-07:
I found a particular case where this results in a large over-estimation of delay.

We have a function where the arguments are provided as an array of bits: `terms: (bits[32])[4]`.  Inside the code, they are directly accessed with compile-time constant index: `terms[0]`, `terms[1]`, etc.   Each of these accesses results in an `array_index` node with literal constant index, but the delay of each is calculated as a dynamic indexing operation.

In reality there is no array -- these are just direct-wired inputs, with the array construct making it easy to parameterize for different numbers of input terms.

With added support for the `HAS_LITERAL_OPERAND`, we'd expect to measure essentially zero delay when the index is a constant.  As it is now, we use the measured delay for a dynamic index even in cases where the array index is a constant.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `SpecializationKind.HAS_LITERAL_OPERAND` is implemented in `xls/estimators/estimator_model.proto:41` and `xls/estimators/estimator_model.py:951,1017,1060`, and populated across `xls/estimators/delay_model/op_models.textproto`, `xls/estimators/delay_model/models/asap7.textproto`, and `xls/estimators/delay_model/models/sky130.textproto` (tested in `xls/estimators/estimator_model_test.py:408-478`). (Closed by Gemini)

closed 2026-11-14 by allight