google/xls #995: add DSLX sample for Dictionary Based Encoding (DBE) encoder/decoder

proppy, 2023-05-30:
See https://en.wikipedia.org/wiki/Dictionary_coder

This will help demonstrate how to implement streaming algorithms that heavily rely on stateful memory with DSLX procs.

proppy, 2023-05-30:
Possible algorithms researched:
- [Adaptive Lossless Data Compression Algorithm](https://www.ecma-international.org/wp-content/uploads/ECMA-222_1st_edition_june_1995.pdf) (ALDC) w/ known hardware implementations ([1](https://ieeexplore.ieee.org/document/5389240), [2](https://ieeexplore.ieee.org/document/10019048))
- https://github.com/google/brotli
- https://facebook.github.io/zstd/

rdob-ant, 2023-05-31:
In addition to the above, we're also considering two other algorithms:
- **DEFLATE** - very commonly used (e.g. in .zip / .gz file formats), based on LZ77-like dictionary coder used together with Huffman encoding.
- **LZ4** - based on pure LZ77-like dictionary coder. Commonly used where especially fast decompression is required (that comes at the expense of reduced compression ratio compared to DEFLATE/Brotli/Zstd).

For me, it is tempting to go with LZ4. Due to its relatively widespread use it would make a much more "useful" DSLX sample, while still being similar to the initially considered ALDC (so most of our ALDC-related ideas are still applicable), and at the same time not resulting in much scope creep (e.g. adding Huffman or entropy coders/decoders on top of the dictionary coder - which would be required for most other algorithms).

proppy, 2023-06-12:
Also learned about the following today:
- https://sneller.io/blog/decompressing-at-over-10-gigabytes-per-second/
- https://github.com/inikep/lizard
- https://github.com/SnellerInc/sneller/tree/master/ion/zion/iguana

rdob-ant, 2023-06-20:
Opened draft #1038 which shows current state of matters