google/xls #998: add syntax sugar for parametric proc instanciation

proppy, 2023-05-30:
Currently one could instanciate parametric structs with `type` definition, see:
https://github.com/google/xls/blob/66230577cddbbb84ba49ac5ff067a5060c2b6a90/xls/examples/delay.x#L202-L204

It would be nice to have something similar for `proc` as wrapping them in a concrete implementation can be quite verbose, see:
https://github.com/google/xls/blob/66230577cddbbb84ba49ac5ff067a5060c2b6a90/xls/examples/delay.x#L206-L227

mtdudek, 2023-05-30:
I think `fn` should also have this feature, just to keep it consistent.

mtdudek, 2023-05-30:
With this syntax sugar, should we handle ` type DelayRamResp32x2048 = RamResp<64>; ` as a valid proc for IR conversion?
Right now procs are a little bit confusing.
Lets us this as an example:
```
 pub proc Delay32x2048_init3 { 
     data_in: chan<u32> in; 
     data_out: chan<u32> out; 
     ram_req: chan<DelayRamReq32x2048> out; 
     ram_resp: chan<DelayRamResp32x2048> in; 
     ram_wr_comp: chan<()> in; 
  
     init { () } 
  
     config(data_in: chan<u32> in, data_out: chan<u32> out, 
            ram_req: chan<DelayRamReq32x2048> out, 
            ram_resp: chan<DelayRamResp32x2048> in, 
            ram_wr_comp: chan<()> in) { 
         spawn Delay<u32:32, u32:2048, u32:3>( 
             data_in, data_out, ram_req, ram_resp, ram_wr_comp); 
         (data_in, data_out, ram_req, ram_resp, ram_wr_comp) 
     } 
  
     next(tok: token, state: ()) { 
         () 
     } 
 } 
```
On one hand it defines new type `Delay32x2048_init3` that inherits from `proc` and overrides `init`, `next` and `next`.
On the other hand if no one calls spawn of it, it becomes an object if its type.

So how should `type DelayRamReq32x2048 = RamReq<10, 64, 0>; ` be handled?
Handling it just as a new type shouldn't be difficult,
but if we expect handling to work as it does with procs.
How should we handle situation where we use both `type DelayRamReq32x2048 = RamReq<10, 64, 0>; `
and `spawn RamReq<10, 64, 0>();`. 
Will DelayRamReq32x2048 become an IR proc, or should it be considered a callee?

Similar question can be raised for `fn`s.

What if we would use 
```
pub proc Delay32x2048_init3 = Delay<u32:32, u32:2048, u32:3> {}
```
or
```
pub Delay<u32:32, u32:2048, u32:3> Delay32x2048_init3 {}
```
I think it would be clearer, as: 
* first one can be read as "I want a public proc  Delay32x2048_init3 that is equal to Delay<u32:32, u32:2048, u32:3> without overrides"
* second is more like: "I want type Delay<u32:32, u32:2048, u32:3> specialization named  Delay32x2048_init3 without overrides"
 
What are your thoughts?

meheff, 2023-05-31:
Related issue which might cover the desired use case: https://github.com/google/xls/issues/890

proppy, 2023-05-31:
thanks @meheff totally missed this, closing as a duplicate sorry!

closed 2023-05-31 by proppy