google/xls #1044: Codegen: bug: failure for a proc with two RAMs that are accessed conditionally

rdob-ant, 2023-06-22:
It is easy to write a proc which has two 1R1W RAMs and accesses them only when certain condition is met (using `send_if`, `recv_if`), that, if codegen is passed `ram_configurations` option, will fail Verilog code generation with a following error:
```
F0622 17:57:02.261463       2 codegen_main.cc:330] Check failed: ::absl::OkStatus() == (xls::RealMain(ir_path)) (OK vs. INVALID_ARGUMENT: Highest value in range fails condition of binary search.)
```

Initially I thought that this was a case of #1043. However, the solution described there, which is to increase number of pipeline stages, does not help here (even with e.g. 1000 stages the codegen still fails). The generation succeeds if we do not ask codegen to rewrite RAM interfaces - that is, it can generate Verilog with classic `data, vld, rdy` streams, but not with `rd_en, wr_en, rd_addr, rd_data, wr_addr, wr_data` buses.

From what I see, the issue happens when accesses to RAM are conditional. If the RAM is always accessed, e.g. you set `let cond = true;`, the codegen succeeds. The generation also succeeds when only one RAM is used and e.g. the read is unconditional but the write is conditional. Although most probably this depends on a specific datapath design, so it may be difficult to generalize based on these isolated observations..

@proppy @cdleary Please take a look at this if possible. Perhaps I'm missing something obvious here and it can be made to work with some extra codegen option? LZ4 encoder from #995 uses two RAMs in a way somewhat similar to what's described here and runs into this issue, which prevents us from generating Verilog module for it with SRAM-like memory interfaces.

---

DSLX proc to reproduce the issue:
```
import xls.examples.ram as ram

type RamWriteReq = ram::WriteReq;
type RamWriteResp = ram::WriteResp;
type RamReadReq = ram::ReadReq;
type RamReadResp = ram::ReadResp;

proc dpram {
    o_write_req1: chan<ram::WriteReq<16, 8, 8>> out;
    i_write_resp1: chan<ram::WriteResp> in;
    o_read_req1: chan<ram::ReadReq<16, 8>> out;
    i_read_resp1: chan<ram::ReadResp<8>> in;

    o_write_req2: chan<ram::WriteReq<16, 8, 8>> out;
    i_write_resp2: chan<ram::WriteResp> in;
    o_read_req2: chan<ram::ReadReq<16, 8>> out;
    i_read_resp2: chan<ram::ReadResp<8>> in;

    init {u16:0}

    config(
        o_write_req1: chan<ram::WriteReq<16, 8, 8>> out,
        i_write_resp1: chan<ram::WriteResp> in,
        o_read_req1: chan<ram::ReadReq<16, 8>> out,
        i_read_resp1: chan<ram::ReadResp<8>> in,
        o_write_req2: chan<ram::WriteReq<16, 8, 8>> out,
        i_write_resp2: chan<ram::WriteResp> in,
        o_read_req2: chan<ram::ReadReq<16, 8>> out,
        i_read_resp2: chan<ram::ReadResp<8>> in,
    ) {
        (
            o_write_req1, i_write_resp1,
            o_read_req1, i_read_resp1,
            o_write_req2, i_write_resp2,
            o_read_req2, i_read_resp2,
        )
    }

    next (tok: token, state: u16) {
        let addr = state;

        // - D=RAM1[addr]
        // - D+=1
        // - RAM2[addr]=D
        // - addr++
        // - if D > 100:
        // -    D=RAM2[addr]
        // -    D+=2
        // -    RAM1[addr]=D
        // -    addr++
        let tok = send(tok, o_read_req1, ram::ReadWordReq<u32:8>(addr));
        let (tok, resp) = recv(tok, i_read_resp1);
        let d = resp.data;
        //
        let d = d + u8:1;
        // 
        let tok = send(tok, o_write_req2, ram::WriteWordReq<u32:8>(addr, d));
        let (tok, _) = recv(tok, i_write_resp2);
        //
        let addr = addr + u16:1;
        //
        // Calculate condition that controls whether below actions are executed
        let cond = d > u8:100;
        //
        let tok = send_if(tok, o_read_req2, cond, ram::ReadWordReq<u32:8>(addr));
        let (tok, resp) = recv_if(tok, i_read_resp2, cond, RamReadResp{
            data: u8:0,
        });
        let d = resp.data;
        //
        let d = d + u8:2;
        // 
        let tok = send_if(tok, o_write_req1, cond, ram::WriteWordReq<u32:8>(addr, d));
        let (tok, _) = recv_if(tok, i_write_resp1, cond, RamWriteResp{});
        //
        let addr = if cond { addr + u16:1 } else { addr };

        addr
    }
}
```

Bazel rules:
```
xls_dslx_library(
    name = "dpram_dslx",
    srcs = [
        "dpram.x"
    ],
    deps = [
        "//xls/examples:ram_dslx",
    ],
)

xls_dslx_test(
    name = "dpram_dslx_test",
    dslx_test_args = {
        "compare": "none",
    },
    library = "dpram_dslx",
)

xls_dslx_ir(
    name = "dpram_ir",
    dslx_top = "dpram",
    library = "dpram_dslx",
    ir_file = "dpram_ir.ir",
)

xls_ir_opt_ir(
    name = "dpram_opt_ir",
    src = "dpram_ir.ir",
    top = "__dpram__dpram_0_next",
)

xls_ir_verilog(
    name = "dpram_verilog",
    src = "dpram_opt_ir.opt.ir",
    verilog_file = "dpram.v",
    codegen_args = {
        "module_name": "dpram",
        "delay_model": "unit",
        "pipeline_stages": "1000",
        "reset": "rst",
        "use_system_verilog": "false",
        "ram_configurations": ",".join([
            "ram1:1R1W:{rd_req}:{rd_resp}:{wr_req}:{wr_resp}".format(
                rd_req = "dpram__o_read_req1",
                rd_resp = "dpram__i_read_resp1",
                wr_req = "dpram__o_write_req1",
                wr_resp = "dpram__i_write_resp1",
            ),
            "ram2:1R1W:{rd_req}:{rd_resp}:{wr_req}:{wr_resp}".format(
                rd_req = "dpram__o_read_req2",
                rd_resp = "dpram__i_read_resp2",
                wr_req = "dpram__o_write_req2",
                wr_resp = "dpram__i_write_resp2",
            ),
        ]),
    },
)
```

grebe, 2023-06-22:
I think what's happening is that there's a state backedge that's longer than one cycle. In this case `addr` is the state element. When `cond = true`, the memory accesses do not constrain the update to state, so there is no long state backedge. However, when cond is dependent on a memory access, this constrains the backedge to have at least the latency of the memory access(es).

I think the compiler may be seeing the intermediate updates to addr as constraining the backedge to include all of the accesses.

I'm curious if the following change to your pseudocode will schedule:
```
        // - D=RAM1[addr]
        // - D+=1
        // - addr++
        // - next_addr = addr
        // - if D > 100:
        //    next_addr++
        // - RAM2[addr-1]=D
        // - if D > 100:
        // -    D=RAM2[addr]
        // -    D+=2
        // -    RAM1[addr]=D
        // -  next_addr

```

The idea is to do the next addr computation as soon as D is available. One access w/ latency=1 should schedule fine, the problem is when there are >1 cycles of latency.

One way to test if a long state backedge including memory accesses is the cause of a scheduling problem is to set the memory latency to 0 (ram_configurations take an optional extra field at the end, e.g. `ram1:RAM_1R1W:{rd_req}:{rd_resp}:{wr_req}:{wr_resp}:{latency}`). If it schedules when memory latency is 0 but fails to schedule otherwise, it's likely because state backedges are constrained by multiple memory accesses.

I'll also add that the RAM support is nascent and there are annoyingly easy ways to break it. #1019 is an example issue (essentially `x=ram[addr]; if (f(x)) x = ram[addr+1];`). I'm actively working on these issues, so things should be improving :)

rdob-ant, 2023-06-23:
Hi @grebe 

I think you're right about the backedge latency being the problem. I've tried a few things:
1. Setting RAM latency to 0 for my original code - codegen succeeds.
2. The pseudocode that you've proposed above will not succeed with RAM latency >=1 - but that's because the next state -  `next_addr` - still depends on the result of RAM read.
3. However, the following, even further simplified single-RAM pseudocode, illustrates the issue:

```
// - addr = state
// - D=RAM1[addr]
// - addr+=1
// - D+=1
// - RAM1[addr]=D
// - if D > 100:
// -    addr++
// - next_state = addr
```

The update of the state depends on value of `D` which is read from RAM. This code **fails** codegen. If you make the state calculation independent of RAM requests, the codegen succeeds:

```
// - addr = state
// - D=RAM1[addr]
// - addr+=1
// - D+=1
// - RAM1[addr]=D
// - if true:
// -     addr++
// - next_state = addr
```

If you want to experiment with it, here's the DSLX:
```
proc dpram {
    o_write_req1: chan<ram::WriteReq<16, 8, 8>> out;
    i_write_resp1: chan<ram::WriteResp> in;
    o_read_req1: chan<ram::ReadReq<16, 8>> out;
    i_read_resp1: chan<ram::ReadResp<8>> in;

    init {u16:0}

    config(
        o_write_req1: chan<ram::WriteReq<16, 8, 8>> out,
        i_write_resp1: chan<ram::WriteResp> in,
        o_read_req1: chan<ram::ReadReq<16, 8>> out,
        i_read_resp1: chan<ram::ReadResp<8>> in,
    ) {
        (
            o_write_req1, i_write_resp1,
            o_read_req1, i_read_resp1,
        )
    }

    next (tok: token, state: u16) {
        let addr = state;

        // - addr = state
        // - D=RAM1[addr]
        // - addr+=1
        // - D+=1
        // - RAM1[addr]=D
        // - if D > 100:
        // -    addr++
        // - next_state = addr
        let tok = send(tok, o_read_req1, ram::ReadWordReq<u32:8>(addr));
        let (tok, resp) = recv(tok, i_read_resp1);
        let d = resp.data;
        //
        let addr = addr + u16:1;
        let d = d + u8:1;
        // 
        let tok = send(tok, o_write_req1, ram::WriteWordReq<u32:8>(addr, d));
        let (tok, _) = recv(tok, i_write_resp1);
        //

        let cond = d > u8:100;
        let addr = if cond { addr + u16:1 } else { addr };

        addr
    }
}
```

and Bazel (the rest is the same as in the issue desc) :
```
xls_ir_verilog(
    name = "dpram_verilog",
    src = "dpram_opt_ir.opt.ir",
    verilog_file = "dpram.v",
    codegen_args = {
        "module_name": "dpram",
        "delay_model": "unit",
        "pipeline_stages": "1000",
        "reset": "rst",
        "use_system_verilog": "false",
        "ram_configurations": ",".join([
            "ram1:1R1W:{rd_req}:{rd_resp}:{wr_req}:{wr_resp}:1".format(
                rd_req = "dpram__o_read_req1",
                rd_resp = "dpram__i_read_resp1",
                wr_req = "dpram__o_write_req1",
                wr_resp = "dpram__i_write_resp1",
            ),
        ]),
    },
)
```

rdob-ant, 2023-06-23:
@grebe and one more question: what does it means for XLS to have a RAM with zero latency in terms of read requests? Does it mean that the data is expected to be available the cycle following the one in which `rd_en` was asserted, or does it mean that the data is expected to be available at the end of the cycle in which `rd_en` is asserted, i.e. it requires an async-read RAM? I presume that writes can always be synchronous.

I think that even the option with sync-write / async-read RAM may be fine for now if it unblocks our progress - as for now we only want to simulate the thing, not implement it in ASIC/FPGA. Though I'll still have to check if there can be an issue due to simultaneous writes&reads, as depending on the expected behavior it may lead to creation of a combinational loop between RAM and XLS-generated module.

grebe, 2023-06-23:
Thanks for the follow-up, I'll play around a bit.

RAMs w/ zero latency are like a flop array, the data is expected to be available within the same cycle that the request is made. Internally, it's really just a scheduling constraint telling the pipeline scheduler how many stages apart to put the request and response- 0 means they should be put in the same stage.

rdob-ant, 2023-07-11:
@grebe With a refined understanding of XLS token semantics after your explanation in #1051, I now see that our biggest problem may lay not in the area of RAM interface rewriting. If the rewriting is broken, we can turn it off, and instead use native XLS channel interfaces in Verilog, specify IO constraints to schedule sends before receives, and even modify the algorithm to use a single-port 1RW RAM so that we're not relying on scheduling of write & read requests sent over parallel channels - 1RW RAM receives them over a single channel, so all accesses are totally ordered.

However, we'll still have one problem: in our algorithm of #995, the next state of proc inevitably depends on the data that is read from RAM, and that data always comes with >=1 cycle of delay (due to a latency between send & recv of >= 1 cycles set by an IO constraint). My experiments show that with such setup, codegen exits with an error, most probably due to a long state backedge.

Do you know how realistic is it to make codegen support such long state backedges? If that is not possible, I'm thinking of one (ugly) potential solution: we can have our algorithm FSM execute `send(ram_req)` and `recv(ram_resp)` in different states, which means that they will be executed in two consecutive iterations of `next()` function. I think that may help us overcome long backedge problem (afaik, we won't even need to use IO constraints with such setup) :thinking: . Do you think this makes sense, and is it really the only way?

ericastor, 2023-07-11:
Funny you should ask - that's actually in flight! codegen currently (as of [one week ago](https://github.com/google/xls/commit/01c595a269a231bec2355576b002fd88c48556ff)) supports long state backedges now.

However, since long state backedges come with throughput implications (if we have a backedge from stage N+K to stage N, we can't achieve worst-case throughput faster than once per K+1 cycles), we're currently planning that designers will need to opt into long state backedge support by specifying a worst-case throughput bound. If you'd like to give feedback on that, maybe we should have that discussion in a separate issue?

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `FindMinimumClockPeriod` and `FindMinimumWorstCaseThroughput` in `xls/scheduling/run_pipeline_schedule.cc:177-187, 215-273` and `Ram1R1WConfiguration::MakeIOConstraints` in `xls/codegen/ram_configuration.cc:180-194` check schedule feasibility before `BinarySearchMinTrue` (avoiding the binary search `CHECK` failure) and support `--worst_case_throughput` for multi-cycle RAM state backedges. (Closed by Gemini)

closed 2026-11-14 by allight