google/xls #1045: Recv_if function incorrectly report errors about init value

prism-pro, 2023-06-26:
I am using a recv_if function in a proc. It should receive a struct from the input channel **x**. 
When I provide the init value in my code `let (tok, data_recv)= recv_if(tok, x, recv_flag,IN_STRUCT{data:constants::IN_TYPE:0,sel:u1:0});` , I get an error:
```bash
0045:         let  recv_flag = SEL == u1:0 || state == u1:1;
0046: 
0047:         let (tok, data_recv)= recv_if(tok,x,recv_flag,IN_STRUCT{data:constants::IN_TYPE:0,sel:u1:0}); // init value is given
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^ ParseError: Expected ')', got ','
0048:         //let (tok, data_recv)= recv_if(tok,x,recv_flag); // init value is not given
0049: 
```
in this way, no IR is generated
then if I remove the init value `let (tok, data_recv)= recv_if(tok,x,recv_flag); ` I get another error:
```bash
0046: 
0047:         //let (tok, data_recv)= recv_if(tok,x,recv_flag,IN_STRUCT{data:constants::IN_TYPE:0,sel:u1:0}); // init revceive value is given
0048:         let (tok, data_recv)= recv_if(tok,x,recv_flag); // init value is not given
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^---------------^ ArgCountMismatchError: Invalid number of arguments passed to 'recv_if'
0049: 
0050:         let delay0_temp = slice(delay0,u32:0, constants::IN_TYPE[5]:[0,...]);
```
But it seems that this error does not affect generating IR and the resulting RTL.

prism-pro, 2023-06-26:
I just tried your code and it make no difference.
```bash
0048: 
0049:         let __default_value = IN_STRUCT{ data: __in_type:0, sel: u1:0 };
0050:         let (tok, data_recv)= recv_if(tok,x,recv_flag,__default_value); // init value is given
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^ ParseError: Expected ')', got ','
0051:         //let (tok, data_recv)= recv_if(tok,x,recv_flag); // init value is not given
0052: 
```
and besides, I am on this commit:
``` bash
commit d2ae71e7d0015582b04babfeb6b1d6a9642c1725 (HEAD -> main, origin/main, origin/HEAD)
Author: Chris Leary <leary@google.com>
Date:   Fri Jun 23 16:08:04 2023 -0700

    [DSLX] Adds regression test case for google/xls#214
    
    Fixes IR conversion of array slice operator that caused a crash.
    
    PiperOrigin-RevId: 542982134
``` 
but the issue exists since I started to use `recv_if` weeks ago. I tried the file `xls/dslx/tests/send_if_recv_if.x` on my machine and got the same:
```
0037: 
0038:     next(tok: token, do_recv: bool) {
0039:         let (tok, foo) = recv_if(tok, r, do_recv, u32:42);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^ ParseError: Expected ')', got ','
0040:         !do_recv
0041:     }
```

mtdudek, 2023-06-26:
I run `bazel clean` and `bazel build -c opt //...` from newest main.
I don't have issues with `xls/dslx/tests/send_if_recv_if.x`.

I run it using `./bazel-bin/xls/dslx/interpreter_main xls/dslx/tests/send_if_recv_if.x` with following results:
```
[ RUN UNITTEST  ] test_main
[            OK ]
[===============] 1 test(s) ran; 0 failed; 0 skipped.
```

cdleary, 2023-06-26:
+1 to what @mtdudek suggested, I think the version of the tools you're using may not be coming from the git hash you cited. Using the `./bazel-bin` path from within your workspace should help ensure you get the right one if you're building from source, and additionally rebuilding as he suggested. (Are you building from source?)

prism-pro, 2023-06-27:
@cdleary @mtdudek  Yes I am building from source.
 Actually after further test, the error is reported by the tool `ir_converter_main` , it is also under the `bazel-bin` directory as shown
```bash
MM:~/xls$ which ir_converter_main
/home/xls/bazel-bin/xls/dslx/ir_converter_main
```
the error occurs when I use `ir_converter_main ./xls/dslx/tests/send_if_recv_if.x` following this [page](https://google.github.io/xls/tools_quick_start/)
```bash
MM:~/xls$ ir_converter_main ./xls/dslx/tests/send_if_recv_if.x
./xls/dslx/tests/send_if_recv_if.x:39:49-39:50
0037: 
0038:     next(tok: token, do_recv: bool) {
0039:         let (tok, foo) = recv_if(tok, r, do_recv, u32:42);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^ ParseError: Expected ')', got ','
0040:         !do_recv
0041:     }
```
and no IR is generated afterwards. So is the function `recv_if` designed as non-synthesizable? Or whether it is just a bug. When I remove the init value of the function, it seem another error will be reported in the test-proc. But if ignore the test-porc, IR can be generated ,and the IR-lowering flow works.

mtdudek, 2023-06-27:
You may find it not amusing, but `./bazel-bin/xls/dslx/ir_convert/ir_converter_main xls/dslx/tests/send_if_recv_if.x` works for me.

prism-pro, 2023-06-27:
> You may find it not amusing, but `./bazel-bin/xls/dslx/ir_convert/ir_converter_main xls/dslx/tests/send_if_recv_if.x` works for me.

Yeah it is not amusing but it helps. I found my `ir_converter_main` is not rebuilt in my last update. Will try again after rebuilding  the project.

prism-pro, 2023-06-28:
@cdleary @mtdudek Thanks a lot. Now the function works for me