google/xls #1067: Bazel dependency on XLS fails to build xls_pip_deps

j2kun, 2023-07-18:
An empty bazel project with XLS as a dependency (minimal repro WORKSPACE file: https://gist.github.com/j2kun/f369293656434bb153d910090b8a2346#file-workspace) cannot build `@xls_pip_deps`

Error message: https://gist.github.com/j2kun/f369293656434bb153d910090b8a2346#file-error-txt

This was tested on XLS at head (`dce44c5c896773c85c2b8078f9d38d7b8b39912e`), but it also fails for the version of XLS that https://github.com/google/fully-homomorphic-encryption was using for a while, `2a613724dd84e891f7536ca81a5e0a960c21b447`. I don't have an exact date when this issue started happening.

j2kun, 2023-07-18:
I got the same issue as above after running `rm -r $(bazel info repository_cache)`

asraa, 2023-07-18:
It seems related to https://github.com/yaml/pyyaml/issues/601

j2kun, 2023-07-19:
Looks like the pyyaml issue is gone, and now I am still working on figuring out the numpy dependency issue. I think it may be because the container I'm using runs on Ubuntu 20.04, and I'm trying upgrading to 22.04.

proppy, 2023-07-24:
Fixed with https://github.com/google/xls/commit/26a8b5cf394165b08150a27f3a29990ca92067c9

closed 2023-07-24 by proppy

Janmajayamall, 2023-08-26:
I happen to be facing same issue on Python 3.10.7 when trying to run https://github.com/google/fully-homomorphic-encryption. Any suggestion on fixing it ?

asraa, 2023-08-28:
> I happen to be facing same issue on Python 3.10.7 when trying to run https://github.com/google/fully-homomorphic-encryption. Any suggestion on fixing it ?

Is the issue still happening if you bump to head to include the XLS update?