google/xls #1071: [DSLX] Feature: Add `#[test_fail]` and `#[test_proc_fail]`

mtdudek, 2023-07-19:
Right now `#[test_proc]` and `#[test]` always has to succeed.

Adding new language testing feature will allow for negative testing.
I think it would only require adding  `#[test_fail]` and `#[test_proc_fail]` to the language parser.
Test procs or fn should keep their current type format, and the bulk of the changes would be
applied to the DSLX interpreter.

cdleary, 2023-07-19:
Related is #481 -- I think a pragma would be non-ideal is we can avoid it vs like, a builtin that traps failure in the bytecode interpreter. Should still be synthesizable in the limit if we think about `fail!` as a fatal error signal.

cunninghamsandwich, 2024-05-02:
Thanks for reporting! As discussed in our April enhancement proposal review meeting: we'll have to think through the syntax we want, so we'll consider this at a lower priority for now.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Expected test failures are supported via the `expected_fail_label` argument on `#[test(...)]` and `#[test_proc(expected_fail_label="...")]` (tested in `xls/dslx/run_routines/run_routines_test.cc:562-575`). (Closed by Gemini)

PR #2109 (2025-07-25): Allow #[test_proc] to expect specific assertion failures
This commit introduces the ability to specify that a test proc is expected to fail on a particular `assert!` or `fail!` in the DSLX interpreter. The `#[test_proc]` attribute accepts an optional `expected_fail` parameter. with the label marking the failure. If a failure occurs with the matching label, the test is treated as successful.

The choice to implement this as an attribute, rather than a built-in, as discussed in https://github.com/google/xls/issues/1071#issuecomment-1642453871 was motivated by cases where there's no clear locatio [... cut]
closed 2026-11-14 by allight