google/xls #1074: codegen: unable to set initiation_interval in DSLX

rdob-ant, 2023-07-20:
Hi,

Since recently the codegen has support for II>1. However, afaik the only way to use that support is by applying `pipeline_init_interval` pragma in XLSCC or using `initiation_interval` attribute in IR source; iow, there is not support for setting II>1 for DSLX code. 

I have DSLX implementation of #995 that requires II=2, and to do that I've locally added an `--initiation_interval` CLI argument to the codegen. The argument is very simple - it just sets II for a top-level proc/function.

Now, we'd like to open a PR with that DSLX code, but I'm not sure what should I do with this new CLI argument. Would anyone like to review a separate PR that adds it, or can we can assume that another approach to setting II in DSLX is already in the works, which means that my II CLI option can temporarily stay in the PR for #995 and we'll have to remove it and rework the code once the "proper" DSLX II support lands? Wdyt @proppy @ericastor ?

proppy, 2023-07-20:
@rdob-ant did you see `--worst_case_throughput`, it got added in https://github.com/google/xls/commit/93b36ba804a4a314ee737d0a4be54cb4378dee35

rdob-ant, 2023-07-20:
@proppy Oh, that's one cryptic name!..

It seems to work, issue resolved. Thanks!

closed 2023-07-20 by rdob-ant

ericastor, 2023-07-20:
Sorry - for a bit of context, we realized that XLS has opportunities around throughput-based optimization that aren't accurately captured by the term "initiation interval" as it's commonly used. Among other things, various HLS systems tend to use II to mean "the throughput should be exactly 1/II"... whereas we may be able to get some additional mileage in certain circumstances by splitting it into two values: "make sure the throughput is **never** worse than this" and "go ahead & save area if it won't slow the throughput to less than this target value".