google/xls #1098: Consider making multiply more explicit WRT overflow

allight, 2023-08-10:
Currently `a:uN * b:uN` is converted to IR `uN = umul[a, b]`. This means that for some values of `a` and `b` overflow is possible.

We might want to consider desuggaring `a*b` to a umul of 2N width and then truncating to N bits. This is more explicit about whats actually going on.

We will need to consider the benefits and costs associated with this.

cdleary, 2023-08-11:
Just a point of clarification -- I think you're saying in the DSLX IR conversion we might consider doing that mapping? But the operator at the DSLX level will always leave the high part unused, so it'll need to immediately slice to get a value of the right width.

Or maybe the proposal is that std::smul/std::umul become builtins that map onto this double-wide IR node instead of widen-then-mul-and-discard-high-bits? I could see that making sense but would be good if we can cite what the builtin will find that the current widen-then-mul misses in practice.

allight, 2023-08-12:
Yeah basically just desugar

`a:U[N] * b:U[N]` to `(std::umul(a, b) as uN)` to make the overflow behavior more explicit.