google/xls #1100: [common] Status matchers in OSS build do not give nice errors

cdleary, 2023-08-15:
"16-byte object" is not a very helpful debug string -- we should figure out how to make this format itself nicely. This was matching with a C string literal via the `StatusIs()` matcher.

```
[ RUN      ] ParserTest.UnterminatedEscapedUnicodeChar
xls/dslx/frontend/parser_test.cc:1858: Failure
Value of: parser.ParseModule()
Expected: has a status code that is equal to 3, and has an error message that is equal to "blah"
  Actual: 16-byte object <41-FD A9-35 3D-56 00-00 00-9B 3D-35 3D-56 00-00>, whose error message is wrong
```

cdleary, 2024-09-10:
Ran into this again today:

```
WARNING: Streamed test output requested. All tests will be run locally, without sharding, one at a time
INFO: Analyzed target //xls/dslx/type_system:typecheck_module_test (0 packages loaded, 5 targets configured).
Note: Google Test filter = TypecheckErrorTest.ArrayEllipsisTypeSmallerThanElements
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TypecheckErrorTest
[ RUN      ] TypecheckErrorTest.ArrayEllipsisTypeSmallerThanElements
xls/dslx/type_system/typecheck_module_test.cc:1458: Failure
Value of: Typecheck("fn main() -> u32[2] { u32[2]:[u32:0, u32:1, u32:0, ...] }")
Expected: has a status code that is equal to 3, and has an error message that has substring "blah"
  Actual: 56-byte object <01-00 00-00 00-00 00-00 10-AF 6E-AE B2-60 00-00 70-74 6E-AE B2-60 00-00 FF-07 20-4A FC-7F 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>, whose status code is wrong
Stack trace:
  0x60b2aceadb55: xls::dslx::(anonymous namespace)::TypecheckErrorTest_ArrayEllipsisTypeSmallerThanElements_Test::TestBody()
  0x72617d43a8fd: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x72617d43a7be: testing::Test::Run()
  0x72617d43bbc1: testing::TestInfo::Run()
... Google Test internal frames ...

[  FAILED  ] TypecheckErrorTest.ArrayEllipsisTypeSmallerThanElements (1 ms)
[----------] 1 test from TypecheckErrorTest (1 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TypecheckErrorTest.ArrayEllipsisTypeSmallerThanElements
````

When this happens I have to make a temporary and log it. @meheff and @grebe I know you have matcher expertise and like the status matchers, do you happen to know if this is an only-in-OSS thing?

mikex-oss, 2024-09-10:
I think this is an OSS thing due to some missing conversion from `absl::Status` to  an error code (`int`) here: https://github.com/google/xls/blob/d8eba72b9c2ba60c210df93691abb196eae35036/xls/common/status/matchers.cc#L50

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/common/status/matchers.h:42,56-93` now delegates `IsOk()`, `IsOkAndHolds()`, and `StatusIs()` directly to Abseil's `::absl_testing::StatusIs` / `IsOkAndHolds` from `absl/status/status_matchers.h`, which includes native `StatusOr<T>` GTest printers instead of printing raw `N-byte object` hex dumps. (Closed by Gemini)

closed 2026-11-14 by allight