google/xls #1101: Wildcard test (`bazel test -c opt ...`) pulls down ASAP7 which is really big

cdleary, 2023-08-15:
@tcal-x is there a target we need to mark as manual? Any way we can bazel query whether this regresses somehow? I think it's happened a few times.

```
~/proj/xls$ bazelisk test -c opt ...
DEBUG: Rule 'org_theopenroadproject' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1659478013 -0700"
DEBUG: Repository org_theopenroadproject instantiated at:
  /home/cdleary/proj/xls/WORKSPACE:69:33: in <toplevel>
  /home/cdleary/proj/xls/dependency_support/initialize_external.bzl:32:33: in initialize_external_repositories
  /home/cdleary/.cache/bazel/_bazel_cdleary/422157be3820d1dd078d709ab2ff14bb/external/rules_hdl/dependency_support/dependency_support.bzl:102:27: in dependency_support
  /home/cdleary/.cache/bazel/_bazel_cdleary/422157be3820d1dd078d709ab2ff14bb/external/rules_hdl/dependency_support/org_theopenroadproject/org_theopenroadproject.bzl:21:10: in org_theopenroadproject
  /home/cdleary/.cache/bazel/_bazel_cdleary/422157be3820d1dd078d709ab2ff14bb/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule git_repository defined at:
  /home/cdleary/.cache/bazel/_bazel_cdleary/422157be3820d1dd078d709ab2ff14bb/external/bazel_tools/tools/build_defs/repo/git.bzl:181:33: in <toplevel>
Analyzing: 2969 targets (298 packages loaded, 45035 targets configured)
    Fetching repository @org_theopenroadproject_asap7; starting 619s
    Fetching https://github.com/The-OpenROAD-Project/asap7/archive/157c92cfd2567a98c5f794fb10c2dd0713516374.tar.gz; 4.3 GiB (4,583,343,777B) 618s
```

cdleary, 2023-08-15:
(We can also maybe discuss the broader strategy we can use here sometime this week -- e.g. maybe we put targets using particular PDKs behind a project-level bazel config that enables them?)

tcal-x, 2023-08-15:
Hmm, the last time this happened was because a `bazel fetch ...` was introduced.   I'll take a look.

closed 2023-08-15 by copybara-service[bot]