google/xls #1119: Confusing error with duplicate procs

meheff, 2023-08-29:
Input:

```
proc foo {
  config() { () }
  init { () }
  next(tok: token, state: ()) { {} }
}

proc foo {
  init { () }
}
```

Error:

```
interpreter_main test.x
test.x:8:3-8:14
0006: 
0007: proc foo {
0008:   init { () }
~~~~~~~~^---------^ ParseError: Module `test` already contains a member named `foo.init` @ test.x:3:3-3:14
0009: }

```

Probably should error out on the duplicate proc declarations.

cdleary, 2023-08-29:
Note this would also be naturally fixed if we had #1029

allight, 2026-11-14:
Keeping this issue open (reviewed by @dplassgit): We are not closing this issue because the confusing collision error still happens since we still create top-level functions for each proc function (even though `Parser::ParseProcLike` in `xls/dslx/frontend/parser.cc` passes `MakeModuleTopCollisionError` to `module_->AddTop(...)` for top-level proc names). (Summarized by Gemini)