google/xls #1135: Support wide bitvector types (>64 bits) in cpp transpiler

meheff, 2023-09-25:
Currently the CppTranspiler (which converts DSLX types to C++ types) generates unusable C++ code for >64 bit types. It compiles but will not give the right result. It is actually convenient that it _does_ compile rather than throw an error because it enables converting a DSLX file which contains wide types into c++ and still use the <= 64 bit types. We should fix this. Probably the representation to use is a std::array of uint8_t. To match the JIT it should use a power of 2 number of bytes.

sandwichmaker, 2023-09-26:
If you are willing to depend on clang intrinsics then perhaps the _BitInt (formerly known as _ExtInt) type maybe easier/better to use?

https://blog.llvm.org/2020/04/the-new-clang-extint-feature-provides.html
https://clang.llvm.org/docs/LanguageExtensions.html#extended-integer-types

meheff, 2023-09-26:
Cool. Thanks for the tip. Didn't know they surfaced the arbitrary width types as intrinsics. In the JIT we used to use the arbitrary width types but the fuzzer kept finding LLVM optimization bugs. Evidently its easy to forget about weird types, especially tiny types, when writing optimizations.  So we abandoned precise-width types and limited ourselves to power-of-two types.  Using _BitInt in the CppTranspiler generated code would be an easy option for support when built with LLVM, but it's probably not too bad to implement it portably using std::array<uint8_t> since we don't actually do any arithmetic in the generated code.

lpawelcz, 2023-11-07:
@proppy Similarly as in https://github.com/google/xls/issues/445 we would find support for types larger than 64bit very useful. 
This issue blocks us in https://github.com/google/xls/pull/1167 and https://github.com/google/xls/pull/1168.
Currently, generation of C++ sources from our [Buffer](https://github.com/google/xls/pull/1167/files#diff-2b3dd20a606f808294f10a32e0dfb91ffcbd6eef0dda2d0270915de501a513c3R27-R30) type specialized for `BSIZE>64` gives incorrect result.
For example, with `Buffer` defined like so:
```
pub struct Buffer {
    contents: bits[128],
    length: u32,
}
```

we get C++ definition:

```
struct Buffer {                                                        
  uint64_t contents;          
  uint32_t length;                                      
                                                                                                                                              
  static constexpr int64_t kContentsWidth = 128;                                                                                              
  static constexpr int64_t kLengthWidth = 32;
                                                                       
  static absl::StatusOr<Buffer> FromValue(const ::xls::Value& value);                                                                         
  absl::StatusOr<::xls::Value> ToValue() const;                        
  std::string ToString(int indent = 0) const;
  std::string ToDslxString(int indent = 0) const;
  absl::Status Verify() const;
  bool operator==(const Buffer& other) const;
  bool operator!=(const Buffer& other) const { return !(*this == other); }
  friend std::ostream& operator<<(std::ostream& os, const Buffer& data);
};
```

Which is of course wrong for the `contents` field.
However, we believe that the usage of `cpp_transpiler` would greatly improve writing C++ tests for Zstandard codec and we look forward to hearing your opinion.

closed 2026-01-30 by copybara-service[bot]