google/xls #1142: Port fuzzer to FuzzTest infrastructure

meheff, 2023-10-04:
To get coverage-guided fuzzing we should port our fuzzer to use https://github.com/google/fuzztest. This would be a significant task. Most of the work would involve changing our ast generator to comply with a fuzztest domain: 

https://github.com/google/fuzztest/blob/ab39c208420d12b11581b82a1573158b1c215b74/doc/team/implementation.md#domains

Basically we need two mechanisms:

1.  Generate a random sample in the domain. This is basically what our existing ast generator does.

2.  Mutate an existing sample potentially with a constraint that the size of the sample should be strictly reduced. We do not have this capability now. Example mutations might be: add operation to function, change return value, replace value with constant, replace value with it's operand, replace value with another in the graph, remove dead expression, and others. The IR minimizer has some of these mutations for reducing IR.

meheff, 2023-10-04:
@ericastor FYI

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While FuzzTest domains were added for IR (`xls/fuzzer/ir_fuzzer/ir_fuzz_domain.h`) and Verilog (`xls/fuzzer/verilog_fuzzer/verilog_fuzz_domain.h`), this issue is specifically about porting the DSLX fuzzer (`AstGenerator` in `xls/fuzzer/ast_generator.cc`) to the FuzzTest infrastructure, so we are not closing this issue. (Summarized by Gemini)