google/xls #1155: BitsView<kBitCount>::GetTypeSize disagrees with JIT for bit counts > 128

allight, 2023-10-17:
The JIT expects bit-values of size 129 <= kBitSize <= 256 to be 64 bytes long. BitsView always expects it to be packed tightly. This can cause asan failures or OOB writes when converting this type of value to a jit buffer.

allight, 2023-10-17:
This is triggered both by the jit code itself and 
 https://github.com/google/xls/blob/d13d1fd88afd91ec2628038edef47cbc469075d4/xls/jit/jit_runtime.cc#L145 disagreeing with the GetTypeSize call that defines how many bytes need to be allocated to the jit value.

allight, 2023-10-17:
Looking even closer this is something that will probably require code-gening view wrappers for each individual type to get the offsets and alignments right. We probably just want to remove the unpacked-views for now since they aren't safe in any but the most basic usages.

cdleary, 2023-10-31:
@allight @meheff Can we give an update on what we think we're going to do in the short vs long term for this? Think we discussed it but would be good to write down the plan.

meheff, 2023-11-01:
@allight Are there static checks we can add to the code which will blow up potentially dangerous types are used?