google/xls #1170: Generating Verilog from designs using DSLX float32 functions

gabizon103, 2023-10-26:
Hi! I'm trying to generate RTL from DSLX, specifically using some of the functions in `float32.x`. To test it out, I wrote a simple program in a file `fmul.x`:
```
import float32

type F32 = float32::F32;

pub fn fmul(
    a: F32,
    b: F32
) -> F32 {
    float32::mul(a,b)
}
```
along with a simple test that passes when using the interpreter:
```
#[test]
fn test0() {
    let a:F32 = F32 {sign: u1:0, bexp: u8:126, fraction: u23:0};
    let b:F32 = F32 {sign: u1:0, bexp: u8:126, fraction: u23:0};
    let exp:F32 = F32 {sign: u1:0, bexp: u8:125, fraction: u23:0};
    assert_eq(fmul(a,b), exp)
}
```

Now, I want to generate IR. I'm running:
`./bazel-bin/xls/dslx/ir_convert/ir_converter_main fmul.x > fmul.ir`, which does output IR. Then, to try to generate Verilog, I use:
`./bazel-bin/xls/tools/codegen_main fmul.ir --top __fmul__fmul --generator=pipeline --delay_model="unit" --output_verilog_path="fmul.v" \
    --module_name=fmul --pipeline_stages=3 --use_system_verilog`

I get this error message: `Error: INTERNAL: Only FFI invocations expected; __float32__mul is not.`
I feel that I am missing something, so any clarifications would be greatly appreciated! Thank you so much!

hzeller, 2023-10-26:
You probably also need to first run the IR through the optimizer:

```
XLSBUILD=bazel-bin/xls
${XLSBUILD}/dslx/ir_convert/ir_converter_main /tmp/fmul.x > /tmp/fmul.ir
${XLSBUILD}/tools/opt_main --top=__fmul__fmul /tmp/fmul.ir > /tmp/fmul.opt.ir
${XLSBUILD}/tools/codegen_main --module_name=fmul --generator=pipeline --delay_model=unit --output_verilog_path=/tmp/fmul.sv --module_name=fmul --pipeline_stages=3 --use_system_verilog /tmp/fmul.opt.ir
```

hzeller, 2023-10-26:
I've updated the error message in that case to be more useful: it points out that the probable cause is the need to run the IR through the optimizer.

gabizon103, 2023-10-27:
Wonderful, this worked! Thanks so much for your help!

cdleary, 2023-10-31:
Also just another resource -- though as I wrote this up I realized it's not the easiest to follow down the rabbit hole -- the Starlark (the Python build DSL) build rules are ultimately wrappers around shell commands, so if you're curious how a target works that you can see in the repo you can usually go and see what commands it desugars into. Some tracing through of the xls_dslx_verilog build rule:

https://github.com/google/xls/blob/15a7e5b6016d75df19b824b599e119630fb08d09/xls/build_rules/xls_rules.bzl#L270

https://github.com/google/xls/blob/15a7e5b6016d75df19b824b599e119630fb08d09/xls/build_rules/xls_rules.bzl#L64

https://github.com/google/xls/blob/15a7e5b6016d75df19b824b599e119630fb08d09/xls/build_rules/xls_ir_rules.bzl#L663

https://github.com/google/xls/blob/15a7e5b6016d75df19b824b599e119630fb08d09/xls/build_rules/xls_ir_rules.bzl#L252

Thanks for filing, we should probably have some additional "flows on the command line" examples in our docs for when the build rules aren't in use.

closed 2023-10-31 by cdleary