google/xls #1171: Combining width-slice with negative offsets.

hzeller, 2023-10-26:
Combining a negative starting index in bit-slice operations with a width-slice does not work as expected, but also does not create an compile error.

Last assert shows the desired operation.
```rust
let a = u3:0b100;
// Get the left-most bit in three ways
assert_eq(u1:1, a[-1:]);    // Control: Left-most bit to the end. Works as expected.
assert_eq(u1:1, a[2+:u1]);  // Starting from left-most 3rd bit, width-slice one bit. Works.
assert_eq(u1:1, a[-1+:u1]); // Starting from left-most bit, width-slice one. Does NOT work.
```

cdleary, 2024-02-22:
This one is actually intended (and tested): https://github.com/google/xls/blob/887387791ae1424c0318472743801c2eca321ffb/xls/dslx/type_system/typecheck_module_test.cc#L1256 But I think it's correct to say that we should give a compile time error when it's a negative constexpr value (e.g. when it's a negative literal number).

For the design space consideration: width slice is different from a normal slice expression because of referential transparency. Consider the following example:

```
x[-2:-1]
```

We can only do that and get a static type out because `-2` and `-1` are both literals (or at least constexpr), so we can compute the size of the result at compile time (type inference time).

However, with a width slice, we can use either a constexpr value:

```
x[8 +: u1]
```

Or we can use a dynamic start index value:

```
x[i +: u1]  // maybe i is 8 dynamically
```

Now consider that `i` can be signed, or it can be unsigned. Right now signed `i` is a type error, because we're not going to create hardware to interpret negative numbers as an "MSb slice" at execution time. As a result, we currently treat whatever is in that position as unsigned, including the literal number `-1`. But it'd be better to flag a negative untyped literal as an error.

cdleary, 2024-02-22:
Sorry I think I buried the lede on that last post -- if `x[-3 +: u3]` worked differently from `let i = s32:-3; x[i +: u3]` it'd be violating referential transparency, binding stuff to a value shouldn't change observable behavior, which is why it's always unsigned and signed values are outlawed in that position.

closed 2024-02-22 by copybara-service[bot]