google/xls #1174: Build downloads two different external python binaries

hzeller, 2023-10-28:
The build downloads two different external binary versions of Python
  * external/python39_x86_64-unknown-linux-gnu/bin/python3.9
  * external/rules_hdl_cpython/install/bin/python3.8
 
Maybe not too much of an issue in reality but probably something to be aware of. Would of course be cool if the different parts could settle one one version.

The escape hatch to use the system Python does now work anymore [`--repo_env=PYTHON_BIN_PATH`](https://github.com/google/xls/blob/main/README.md?plain=1#L112-L114), so probably worthwhile removing that documentation.

cdleary, 2023-11-01:
Is this because we need to update rules_hdl to be consistent with the XLS repo? cc @mithro who had been looking at that

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): XLS migrated to Bzlmod (`MODULE.bazel:12-38`) using `rules_python`'s hermetic `python.toolchain(python_version = "3.14")` and `pip.parse`, and removed `PYTHON_BIN_PATH` from `README.md`, eliminating the legacy `WORKSPACE` dual-Python setup. (Closed by Gemini)

closed 2026-11-14 by allight