google/xls #1191: DSLX: bug: IR build error dependent on order of `let` and `trace_fmt`

mczyz-antmicro, 2023-11-10:
I experienced an error in a rather simple DSLX function execution, which is dependent on order of expressions: `let` and `trace_fmt`.
XLS build information: from commit 16ace44495711715f505ad231bcd2cb74b04a9a9 (November 3rd)

The only difference between 2 functions is the order, in which call to `trace_fmt!` is executed:

```
pub fn myFunctionOK() {
    let crc: u32 = u32:0;
    for (i, crc): (u32, u32) in range(u32:0, u32:8) {
        trace_fmt!("crc = {}", crc); // Seems fine
        let crc = crc + i;
      (crc)
    }(crc);
}

pub fn myFunctionNotOK() {
    let crc: u32 = u32:0;
    for (i, crc): (u32, u32) in range(u32:0, u32:8) {
        let crc = crc + i;
        trace_fmt!("crc = {}", crc); // Causes failures?
      (crc)
    }(crc);
}
```

If you try to interpret function `myFunctionNotOK` you get an error:

`Error: INVALID_ARGUMENT: Could not build IR: Return type of function ____itok__random__random_counted_for_0_body used as counted_for body should have (token, bits[1], bits[32]) type, got (token, bits[1], token) instead: counted_for.16: (token, bits[1], bits[32]) = counted_for(tuple.15, trip_count=8, stride=1, body=____itok__random__random_counted_for_0_body, id=16)`

Runs are successful with option `--compare none`, which I believe turns the IR step off and the error is not raised.

I attach:
* pass.log - logs from run with option `--compare none`
* fail.log - logs from failed run
* random.x.log - full test case to reproduce
[random.x.log](https://github.com/google/xls/files/13320386/random.x.log)
[pass.log](https://github.com/google/xls/files/13320387/pass.log)
[fail.log](https://github.com/google/xls/files/13320388/fail.log)

ericastor, 2023-12-21:
The error here could probably be better - but the issue is that the last statement in the `for`'s body is taken to be the returned accumulator value, as is standard for return values in DSLX.

This might be worth improving, but I believe it is fundamentally working as intended here.

mczyz-antmicro, 2023-12-21:
Can you explain why JIT claims that the return type is (..., token) and not (..., u32) even though `crc` of type u32 is the return statement?

```
pub fn myFunctionNotOK() {
    let crc: u32 = u32:0;
    for (i, crc): (u32, u32) in range(u32:0, u32:8) {
        let crc = crc + i;
        trace_fmt!("crc = {}", crc); // Causes failures?
        crc // <-- Isn't this statement returned?
    }(crc);
}
```

If I change the loop body to return the `trace_fmt` :

```
for (i, crc): (u32, u32) in range(u32:0, u32:8) {
      let crc = crc + i;
      trace_fmt!("crc = {}", crc)
  }(crc);
```

I do get an error:

`      |________^ XlsTypeError: uN[32] vs token: For-loop init value type did not match for-loop body's result type.`

which I believe is the intended result you mentioned

ericastor, 2023-12-21:
Apologies, I missed the trailing `(crc)` in your original example. I'm not sure why this error is happening, and I'll try to make sure it gets looked at.

commit 465d8552 (2023-12-21): Always identify the last expression in a counted-for body as the "return value"

If implicit token handling was required, this previously grabbed the last value created, even if the body ended with a final expression to signal the intended value.

Fixes #1191

PiperOrigin-RevId: 592909823
closed 2023-12-21 by copybara-service[bot]