google/xls #1196: Continuation of expression removes indentation

hzeller, 2023-11-16:
Consider the following with a long expression that is continued at the next line with appropriate indentation:

```rust
fn foo(some_value: u32) -> u32 {
    some_value + some_value + some_value + some_value + some_value + some_value + some_value +
        some_value
}
```

The formatter removes this indentation of the continuation and starts the continuation in the column the original expression started:

```rust
fn foo(some_value: u32) -> u32 {
    some_value + some_value + some_value + some_value + some_value + some_value + some_value +
    some_value
}
```

I suspect aligning the continuation with the beginning of the original expression is only an issue in free-standing expressions used as a 'return from something' situation. In assignments, the current behavior produces good results as the indentation is given by the assignment in the line above:

```rust
fn foo(some_value: u32) -> u32 {
    let a = some_value + some_value + some_value + some_value + some_value + some_value + some_value +
            some_value;
}
```

cdleary, 2023-11-17:
The let statement is what causes the RHS to be at the alignment of the `= ` -- I'm not sure if it makes sense to align all expressions that overflow a single line one nesting level?

hzeller, 2023-11-20:
I think indentation in continuation of expressions will improve readability.

Compare the first (hand-written) and second (formatter generated) example. The second looks like `foo()` just returns `some_value` as it hides the fact that it is a continuation and only if the reader explicitly looks at the end of the previous line sees that this is actually part of a long expression (we have seen how subtle things at the end of lines can be easily overlooked by readers of code...)

 (that is why I personally also advocate to move the operator that continuates in the new line in other code, but I guess that ship has sailed).

(The `let` statement is perfectly indented and should not change; the LHS forces indentation of everything on the RHS, so that makes clear what is happening)

dplassgit, 2025-01-06:
This is definitely still broken.