google/xls #1199: A problem occurs when using the lec_main tool. How to use the lec_main tool correctly? Is there any example?

yaqianp, 2023-11-20:
The cell name of the netlist generated by using rules_hdl repository cannot match the logic of the NodeToNetlistName function in lec_main. As a result, the verification fails. Because z3 ast is empty, the verification result is always equivalent.

yaqianp, 2023-11-20:
xls ir：a_and_b.ir
`
package my_package

file_number 1 "./loop.cpp"

top fn test_func(a: bits[1], b: bits[1]) -> bits[1] {
  literal.5: bits[1] = literal(value=1, id=5, pos=[(1,3,5)])
  literal.4: bits[1] = literal(value=1, id=4)
  not.6: bits[1] = not(literal.5, id=6, pos=[(1,3,5)])
  not.7: bits[1] = not(literal.5, id=7, pos=[(1,2,1)])
  ret and.3: bits[1] = and(a, b, id=3, pos=[(1,3,5)])
}

`

netlist：a_and_b.v
`
/* Generated by Yosys 0.8+ */

(* hdlname = "\\test_func" *)
(* top =  1  *)
(* src = "test2/xls.v:1.1-7.10" *)
module test_func(a, b, out);
  (* src = "test2/xls.v:2.14-2.15" *)
  wire _0_;
  (* src = "test2/xls.v:3.14-3.15" *)
  wire _1_;
  (* src = "test2/xls.v:4.15-4.18" *)
  wire _2_;
  (* src = "test2/xls.v:2.14-2.15" *)
  input a;
  (* src = "test2/xls.v:3.14-3.15" *)
  input b;
  (* src = "test2/xls.v:4.15-4.18" *)
  output out;
  AND _3_ (
    .A(_1_),
    .B(_0_),
    .Y(_2_)
  );
  assign _1_ = b;
  assign _0_ = a;
  assign out = _2_;
endmodule
`

cells library：cmos_cells.lib
`
// test comment
/* test comment */
library(demo) {
  cell(BUF) {
    area: 6;
    pin(A) { direction: input; }
    pin(Y) { direction: output;
              function: "A"; }
  }
  cell(NOT) {
    area: 3;
    pin(A) { direction: input; }
    pin(Y) { direction: output;
              function: "A'"; }
  }
  cell(NAND) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A*B)'"; }
  }
  cell(AND) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A*B)"; }
  }
  cell(NOR) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A+B)'"; }
  }
  cell(OR) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A+B)"; }
  }
  cell(DFF) {
    area: 18;
    pin(C) { direction: input;
                 clock: true; }
    pin(D) { direction: input; }
    pin(Q) { direction: output;
              function: "IQ"; }
    ff(IQ, IQN) { clocked_on: C;
                  next_state: D; }
  }
  cell(DFFSR) {
    area: 18;
    pin(C) { direction: input;
                 clock: true; }
    pin(D) { direction: input; }
    pin(Q) { direction: output;
              function: "IQ"; }
    pin(S) { direction: input; }
    pin(R) { direction: input; }
    ff("IQ", "IQN") { clocked_on: C;
                  next_state: D;
                      preset: S;
                       clear: R; }
  }
}
`

lec_main command
./lec_main --ir_path=a_and_b.ir --netlist_path=a_and_b.v --cell_lib_path=cmos_cells.lib

cdleary, 2023-11-21:
I think there's not good examples right now because we made the LEC tool before we had the open source PDKs available in the repository -- I'll take a look to see if we can get the LEC flow working on the open source PDKs to create an example flow.