google/xls #1200: Fuzzer generates `decode` + `bit_slice` programs with incorrect JIT results

ericastor, 2023-11-20:
Example: (minimized from crasher 30949b97 [2023-11-19])
```
package sample

top fn __sample__main(x2: bits[8]) -> bits[33] {
  x6: bits[129] = decode(x2, width=129, id=7)
  ret bit_slice.9: bits[33] = bit_slice(x6, start=0, width=33, id=9)
}
```

`./eval_ir_main --input='bits[8]:0x80' --test_llvm_jit sample.ir` shows the error:
```
Error: INVALID_ARGUMENT: Miscompare for input[0] "bits[8]:0x80"
  JIT: bits[33]:0x1
  interpreter: bits[33]:0x0
```
This persists with `--llvm_opt_level=1`, but stops with `--llvm_opt_level=0`, so this seems like some sort of LLVM optimization error.

ericastor, 2023-11-20:
Suspected cause is this diff caused by LLVM's `instcombine` pass; this occurs between `opt` run with `--debug-counter=instcombine-visit-skip=0,instcombine-visit-count=25` and `instcombine-visit-count=26`:

```
--- /tmp/test.instcombine_25.ll	2023-11-20 18:14:53.338085298 +0000
+++ /tmp/test.instcombine_26.ll	2023-11-20 18:14:59.270074629 +0000
@@ -25,7 +25,7 @@
 entry:
   %x2 = load i8, ptr %operand_ptr, align 1
   %0 = zext nneg i8 %x2 to i256
-  %1 = icmp uge i8 %x2, -127
+  %1 = icmp ugt i8 %x2, -128
   %2 = shl i256 1, %0
   %3 = select i1 %1, i256 0, i256 %2
   %4 = and i256 %3, 680564733841876926926749214863536422911
```

This seems like it can't be correct, since -128 isn't representable as an i8.

ericastor, 2023-11-21:
Suspect that additional examples were found as crashers 7eacdf7a and cf9385ef - and maybe 6c14f73c and fe61e995 (though those may be a combined example with issue #1201).

ericastor, 2023-11-21:
Filed as llvm/llvm-project#72911

hzeller, 2024-09-12:
The upstream bug seems to be fixed, we're using a sufficiently recent llvm snapshot and the fuzzer test is passing now (https://github.com/google/xls/commit/9a266ca4e37b75b07891d2895fecaa95922a8ea6).

@ericastor can you confirm (and close), or there other things to test ?

ericastor, 2024-09-12:
That should be enough!

closed 2024-09-12 by ericastor