google/xls #1201: Fuzzer generates `decode` + `zero_ext` + `concat` programs with incorrect JIT results

ericastor, 2023-11-21:
Example: (minimized from crasher 0a236eec [2023-11-19])
```
package sample

top fn __sample__main(x1: bits[8]) -> bits[520] {
  x5: bits[256] = decode(x1, width=256, id=13)
  zero_ext.57: bits[512] = zero_ext(x5, new_bit_count=512, id=57)
  ret concat.67: bits[520] = concat(zero_ext.57, x1, id=67)
}
```

`./eval_ir_main --input='bits[8]:0xff' --test_llvm_jit sample.ir` shows the error:
```
Error: INVALID_ARGUMENT: Miscompare for input[0] "bits[8]:0xff"
  JIT: bits[520]:0xff
  interpreter: bits[520]:0x80_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_0000_00ff
```

This persists with `--llvm_opt_level=1`, but stops with `--llvm_opt_level=0`, so this seems like some sort of LLVM optimization error.

Filed as llvm/llvm-project#72927

ericastor, 2023-11-21:
Suspect that additional examples were found as crashers 2432c176 and 4146047a - and maybe 6c14f73c and fe61e995 (though those may be a combined example with issue #1201).

hzeller, 2024-09-12:
The upstream llvm bug seems to be fixed, we're using a sufficiently recent llvm snapshot and the fuzzer test is passing now (https://github.com/google/xls/commit/9a266ca4e37b75b07891d2895fecaa95922a8ea6).

@ericastor can you confirm (and close), or there other things to test ?

ericastor, 2024-09-12:
That should be enough!

closed 2024-09-12 by ericastor