google/xls #1209: Bug in Dual Port Memory, write on port1 is wired to port0

mczyz-antmicro, 2023-11-29:
Observed issue with the True Dual Port Memory Model in `examples/ram.x`.

Writes to port 1 are performed on port 0 in `examples/ram.x`. In line https://github.com/google/xls/blob/main/xls/examples/ram.x#L660, if a valid write request is received on port1, then memory is updated with data from request to port0. 

```
    // Do writes.
    let state = if valid0 && request0.we {
      update(state, request0.addr, request0.data)
    } else {
      state
    };
    let state = if valid1 && request1.we {
      update(state, request0.addr, request0.data)
    } else {
      state
    };
```

This is incorrect and prevents from using port1 as a write port or using the model for 2 simultaneous writes. The fix is simple and I will open a PR to provide it (part of #1208).

proppy, 2023-11-30:
/cc @grebe

mczyz-antmicro, 2024-01-02:
Closing - #1210 is merged.

closed 2024-01-02 by mczyz-antmicro